IFL-CAMP/easy_handeye #1: Plugin error

animez4me, 2017-08-17:
Hi,

After running Calibrate.launch, I see rviz and the GUI being launched but get the following error,

PluginManager._load_plugin() could not load plugin "rqt_easy_handeye/Hand-eye Calibration automatic movement":

How can I load the plugin?

Thanks,

marcoesposito1988, 2017-08-21:
Hi @animez4me,

did you run catkin_make and source the workspace's devel/setup.bash ? This could be a reason, I think.

animez4me, 2017-08-21:
Hello, yes I did, catkin_make was successful since everything else launched.

marcoesposito1988, 2017-08-28:
I'm sorry, I don't manage to reproduce your problem.

What version of Ubuntu and ROS are you using?

serabuyukgoz, 2018-05-29:
Hi @wmy101,

Could you solve your problem by any chance?

marcoesposito1988, 2018-12-17:
Closing for inactivity. If the problem persists, please comment and the issue will be reopened if appropriate.

iamyk666, 2019-01-23:
hollow, I have meet the same problem, do you solve this problem? @wmy101

aPonza, 2019-07-10:
I had the same issue, I solved it by adding `<exec_depend>rqt_easy_handeye</exec_depend>` in my own package.xml, but I think it should rather go in easy_handeye's own package.xml as exec_depend, since it's used in calibrate.launch.

marcoesposito1988, 2019-07-10:
That's interesting! It sounds like we have a solution to this annoying problem. I'll look into it. Thanks!

nalt, 2019-11-06:
I had to remove those two lines in install(): https://github.com/IFL-CAMP/easy_handeye/blob/bb928fbca7ec7599878e16694610d4c17101075b/rqt_easy_handeye/CMakeLists.txt#L13-L14

This fixed it for me in install space (had no problem in devel space). The plugin got installed at two locations, once by catkin_python_setup(), once by install().

marcoesposito1988, 2019-11-19:
That sounds right. I applied the change, thanks!

closed 2020-01-09 by marcoesposito1988

ZZ-111, 2021-01-16:
> 我遇到了同样的问题，我通过添加`<exec_depend>rqt_easy_handeye</exec_depend>`自己的package.xml来解决了这个问题，但是我认为应该将它作为exec_depend放入easy_handeye自己的package.xml中，因为它已在calibrate.launch中使用。
> 
> 编辑：我发现我的建议存在循环依赖问题。对于我来说，依靠我使用的启动文件还是没有什么意义的，对吧？

Thanks!That's vrey useful!