IFL-CAMP/easy_handeye #2: About eye_base calibration

GPRAMESH, 2017-09-27:
Hello Sir,
I tried to do calibration while keeping kinect on base. I am not able to produce accurate results even though I tried lot of times. Can I know what is the procedure for calibration eye on base rather than eye on hand. I am trying for two months which delays my thesis.

Thanks in advance

marcoesposito1988, 2017-10-06:
Hi @GPRAMESH,

if your camera is attached to the robot hand, it is "eye in hand" (so the eye_on_hand parameter should be set to true); if the camera is not attached to the robot, it is then an "eye on base" calibration (so eye_on_hand:=false).

Please check:
- that the frames are correct (the camera frame is the optical centre of the camera and the robot end effector frame corresponds to the last joint of the robot)
- that you are not including outliers in the samples (for example if your tracking system keeps reporting a wrong position to tf even if the marker is out of sight)

GPRAMESH, 2017-10-06:
for eye_on_base calibration, I kept eye_on_hand:=false. I used kinect package such as iai_kinect2  in order to track markers(apriltags package). I kept apriltags at robot end effector and kinect at some base.Before taking samples from GUI, I kept kinect at 0 0 0 0 0 0 position. I managed to take all samples correct and still I am not able to find transformation from base to kinect frame. Could any one help me how to sort out this one?

GPRAMESH, 2017-10-06:
These are below values I got. When I keep in launch file I am getting wrong values. It is very important for my thesis. Please help me out how to sort this. Thanks in advance
hand to world transformation
--
Sample.No | 1 | 2 | 3 | 4 | 5 | 6 | 7 | 8 | 9 | 10 | 11 | 12 | 13 | 14 | 15
x | -0.009 | -0.303 | 0.009 | -0.455 | 0.086 | -0.094 | -0.251 | -0.093 | -0.326 | -0.184 | -0.189 | -0.185 | -0.235 | -0.140 | -0.184
y | -0.233 | -0.537 | 0.047 | -0.040 | -0.466 | -0.265 | -0.399 | -0.106 | -0.148 | -0.262 | -0.284 | -0.243 | -0.256 | -0.256 | -0.262
z | 0.822 | 0.570 | 0.873 | 0.737 | 0.696 | 0.806 | 0.701 | 0.853 | 0.782 | 0.804 | 0.791 | 0.791 | 0.783 | 0.800 | 0.804
  |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
qx | 0.464 | 0.597 | 0.651 | 0.621 | 0.662 | 0.567 | 0.635 | 0.667 | 0.644 | 0.657 | 0.656 | 0.658 | 0.657 | 0.657 | 0.657
qy | 0.883 | 0.755 | 0.735 | 0.723 | 0.735 | 0.820 | 0.754 | 0.738 | 0.739 | 0.748 | 0.748 | 0.747 | 0.747 | 0.748 | 0.748
qz | 0.017 | 0.266 | -0.162 | 0.059 | 0.036 | 0.039 | 0.155 | -0.056 | 0.058 | 0.054 | 0.058 | 0.053 | 0.051 | 0.052 | 0.054
qw | 0.071 | 0.051 | 0.100 | 0.296 | -0.145 | 0.070 | 0.063 | 0.090 | 0.189 | 0.075 | 0.079 | 0.077 | 0.087 | 0.075 | 0.075

camera to marker transformation
--
Sample.No | 1 | 2 | 3 | 4 | 5 | 6 | 7 | 8 | 9 | 10 | 11 | 12 | 13 | 14 | 15
x | 0.079 | 0.089 | 0.095 | 0.084 | 0.083 | 0.083 | 0.085 | 0.085 | 0.084 | 0.085 | 0.069 | 0.098 | 0.104 | 0.063 | 0.085
y | -0.452 | -0.445 | -0.479 | -0.468 | -0.441 | -0.457 | -0.444 | -0.461 | -0.463 | -0.463 | -0.448 | -0.456 | -0.429 | -0.470 | -0.463
z | 1.129 | 1.148 | 1.159 | 1.152 | 1.125 | 1.141 | 1.134 | 1.134 | 1.152 | 1.125 | 1.122 | 1.137 | 1.099 | 1.162 | 1.125
  |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
qx | 0.974 | 0.899 | 0.891 | 0.959 | 0.901 | 0.959 | 0.940 | 0.910 | 0.943 | 0.930 | 0.930 | 0.963 | 0.934 | 0.929 | 0.930
qy | 0.063 | 0.039 | 0.129 | -0.105 | 0.237 | 0.124 | 0.150 | 0.141 | 0.027 | 0.152 | 0.145 | 0.057 | 0.134 | 0.144 | 0.152
qz | 0.086 | 0.222 | -0.221 | -0.161 | 0.302 | -0.186 | -0.295 | -0.284 | -0.280 | -0.287 | -0.286 | 0.174 | -0.284 | -0.290 | -0.287
qw | -0.200 | -0.377 | -0.374 | -0.210 | -0.203 | -0.176 | -0.085 | -0.266 | -0.179 | -0.171 | -0.180 | -0.198 | -0.173 | -0.180 | -0.171

Kinect location approx. value (before calibration given in launch file)
--
x | 0.88
y | 1.13
z | 0.96
roll | -0.45
pitch | 3.14
yaw | 1.1

Calibration output
--
x | y | z | qx | qy | qz | qw
1.217 | 0.517 | 0.087 | 0.028 | -0.437 | -0.090 | 0.894

marcoesposito1988, 2017-10-09:
Did you check the AR marker tracking? Did you turn on the depth image and check that the tf frame of the marker lies on the point cloud?

If the calibration of the RGB camera is faulty, or if you did not configure the physical size (in centimeters) of the marker correctly, the coordinates returned by the marker tracker would be completely wrong. You can check this against the depth information given by the RGB-D camera.

marcoesposito1988, 2018-12-17:
Closing for inactivity. If the problem persists, please comment and the issue will be reopened if appropriate.

closed 2018-12-17 by marcoesposito1988