IFL-CAMP/easy_handeye #4: Can't compute the transformation from the base frame to the world frame

PhiAbs, 2017-10-05:
I am trying to calibrate a realsense camera using the easy_handeye package. I track some ar markers with the ar_track_alvar package. I want to do a  eye on base calibration. Using the GUI from the easy_handeye package, I managed to take samples. The transformations are shown in the "samples" window and they are correct. However, if I want to compute the transformation between the world frame and the camera, the transformation shown in the "Result" window is always the following: 

translation: 
  x: 0.0
  y: 0.0
  z: 0.0
rotation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 1.0

For the frames I used the following.
Tracking Base Frame:  The camera frame
Tracking marker Frame: The frame of the tracked markers, in the camera system
Robot Base Frame: The world frame
Robot Effector Frame: The robot gripper frame which holds the ar markers

Does anyone know what the issue is or what I did wrong?

marcoesposito1988, 2017-10-06:
Hi @PhiAbs,

could you please post the collected samples?

This package uses ViSP as a backend. Have you tried using it directly, passing the samples collected with easy_handeye? If the result is then correct, we have a problem passing it the results or fetching back the result.

marcoesposito1988, 2018-12-17:
Closing for inactivity. If the problem persists, please comment and the issue will be reopened if appropriate.

closed 2018-12-17 by marcoesposito1988