IFL-CAMP/easy_handeye #8: canTransform  Error

LeonRobot, 2017-12-07:
Hi, I'm using easy_handeye to calibrate my primesense camera with the robot. I got this error "Error processing request: canTransform: source_frame camera_marker does not exist."
Does the camera_marker need to change or need to be added to the TF tree?
Thanks very mush!

marcoesposito1988, 2017-12-11:
Hello @LeonRobot,

as the error message says, this happens because the tf frame you specified doesn't exist in the tree.

When you have your tracking system running, you should have two tf frames, one for the position of the camera (or field generator, ...) and one for the marker (or sensor, ...). You need to override the relative parameters of the launch file with the ID of these frames.

The tracking system should of course be running before you start sampling points.

Hope this can help.

LeonRobot, 2017-12-11:
Thank you very much for your reply!  I have fixed this problem.  As you said,  the tf frame you specified  must exist in the tree.

closed 2017-12-11 by LeonRobot

LeonRobot, 2018-04-10:
What you should do is replace ‘camera_marker’ with the name your camera's link name.
Here are some examples.
    <arg name="camera_frame"          default="camera_rgb_optical_frame"/>
    <arg name="reference_frame"       default="camera_link"/>
I hope this will help you!

Mylittledream, 2018-04-10:
Thanks for your help!