IFL-CAMP/easy_handeye #11: launch file parameters

tymanab, 2018-04-06:
Hi,
I was trying to calibrate a ZED stereo camera with UR5 in eye_in_hand mode and there are a few parameters in launch file that I don't know how to modify.

I tried to modify example.launch and got stucked at these two lines:
```xml
 <arg name="tracking_base_frame" value="camera_link" />
<arg name="tracking_marker_frame" value="camera_marker" />
```
I want to verify what "camera_link" and "camera_marker" refers to and which package/what message are used to publish them when using Kinect.
Thanks in advance for all your help.

marcoesposito1988, 2018-04-10:
Hi @tymanab,

the value of those parameters is the name of the tf frame. If your tracking system publishes the output as a transform from the "camera" to the "marker" tf frames, then you can modify those parameters like:

```xml
<arg name="tracking_base_frame" value="camera" />
<arg name="tracking_marker_frame" value="marker" />
```

marcoesposito1988, 2018-12-17:
Closing for inactivity. If the problem persists, please comment and the issue will be reopened if appropriate.

closed 2018-12-17 by marcoesposito1988