IFL-CAMP/easy_handeye #13: The tf between ur3 and xtion is unfixed!

gaoyaNAN23, 2018-04-27:
Hello~
I'm using your program to calibrate the ur3 and my xtion. And I use the way of eye on base.
I saw the program and I thought the result should be the tf between ur3 and xtion. So it should be a fixed value. But I found , when I add a/several samples , the result changed a lot.
I don't know what's wrong with it. Do you have any idea about my problem? Or can you tell me how can I test if the camera_marker_samples and the hand_world_samples are right?
Thanks a lot!  Look forward to your reply!

marcoesposito1988, 2018-05-02:
Hello @gaoyaNAN23,

yes, the algorithm should compute the transformation between the base of the robot and the optical frame of the camera. So if the robot and the camera are not moving with respect to each other, and you calibrate multiple times, you should always get the same result.

The calibration could fail if:
- the robot movement does not happen about 3 independent axes of rotation (if you are using the automatic movements plugin, this should be covered for you)
- the tf frames you are passing as robot base and camera optical frame are not fixed, but move when you move the robot
- the marker is not solidly fixed to the robot's hand
- the marker tracking is not stable (if you are using AR markers, they are unstable when the marker is perpendicular to the camera's optical axis)
- the size of the marker is wrongly configured in the AR marker tracking software
- the camera is not calibrated (see the camera_calibration ROS package)
- the marker tracking system has some lag, and you sample the point before the marker has stopped moving
- you are not sampling enough points (in my practical experience, you need 7)

There are of course more possible reasons, but these I can think off the top of my head.

marcoesposito1988, 2018-05-29:
Hey @gaoyaNAN23,

did you manage to solve the issue? If the problem was not a problem with the package per se, I would close the issue.

gaoyaNAN23, 2018-05-30:
Hi,@Marco Esposito,
At first,I want to say thanks to you.Your package is so great! What's more ,your answer about my issue is very helpful. 
When I used about 15 sample points , the tf I needed was almost fixed , and it looked like a right one.
Now,I'm trying to get the pose estimation of an object , and then I can test if the tf between the ur and kinect or not.
So I'm sorry to reply you late .I mean to reply you after the testing.
Thanks for your help angin. Best wishes for you !

gaoyaNAN23, 2018-05-30:
Hi,Marco Esposito,
At first,I want to say thanks to you.Your package is so great! What's more ,your answer about my issue is very helpful.
When I used about 15 sample points , the tf I needed was almost fixed , and it looked like a right one.
Now,I'm trying to get the pose estimation of an object , and then I can test if the tf between the ur and xtion or not.
So I'm sorry to reply you late .I meant to reply you after the testing.
Thanks for your help angin. Best wishes for you !

marcoesposito1988, 2018-06-04:
Great to hear! I will then close this issue.
You are welcome to open a new one in case you encounter a problem with the software.

closed 2018-06-04 by marcoesposito1988