IFL-CAMP/easy_handeye #15: Camera frame position keeps on changing frequently

rajmohan747, 2018-05-21:
Hi,
I am using UR5 robot and Zed camera for the calibration purpose.
Is it possible to mount the camera ,inverted in a stand and do the robot-camera calibration with **eye_on_hand** parameter set as **false**.
As of now the **camera frame keeps on changing** its position **very frequently** after publishing transformation found during calibration.
So I just want to know whether I am doing some other thing wrong or whether the package itself is apt for such a camera mounting condition

arjunskumar, 2018-05-23:
I'm getting the same issues. Any solution plz

rajmohan747, 2018-05-23:
Also after the calibration process was done and publishing the transformation, I want the robot to move to a specific point within camera range.
On checking the camera coordinate values (Xc,Yc,Zc) at that point,it was very different from that of robot coordinate values at the very same point(Xr,Yr,Zr).Actually **both the coordinate system** should **share same values for any point in workspace** after calibration ,right?That is for any point in Robot coordinate system (Xr,Yr,Zr) it should be have same value in camera coordinate system(Xc,Yc,Zc).
Please correct me ,if I am wrong

rajmohan747, 2018-05-23:
Also I have assumed that the calibration process is done correctly by keeping the ArUco marker in the robot wrist and moving the same within the camera's reach.During the motion it was observed in Rviz that the Robot' motion and the marker's motion were synchronized.That is ,the marker stays in the robot end effector position throughout the robot motion.
Hence I assumed both robot and camera are sharing a common coordinate system and calibration is done.

marcoesposito1988, 2018-05-29:
Hi @rajmohan747,

I am not sure if I understand you correctly. After performing the calibration, you can run check_calibration.launch to test the outcome. In here you should see the whole system represented in tf: the robot links (provided by the robot_publisher), the transform from the base of the robot to the base of the tracking system (i.e. the camera for an optical tracking system) and the frame attached to the marker.

If the calibration is correct, the transformation between the end effector of the robot and the marker should stay constant (up to an epsilon), regardless of the position of the robot.

Once this is done, your constellation of reference frames is correct, so if you tell the robot to go to a position with respect to any frame (including the camera's), moveit should be able to transform the position into the robot's reference frame, and reach it.

However: you should check that you are explicitly telling moveit to what reference frame your coordinates refer to. Otherwise, if you take the raw input of the tracking system (which is in the camera's reference frame), and you tell the robot to go to those coordinates without specifying the reference frame, it could assume that they are in the robot base's reference frame; so the robot motion would be completely wrong. 

The most error-proof way is to use a PoseStamped as target for the motion of the robot, where you set both the coordinates and the reference frame they refer to. Moveit will take care of transforming them automatically with tf (even if this was bugged once, IIRC).

marcoesposito1988, 2018-12-17:
Closing for inactivity. If the problem persists, please comment and the issue will be reopened if appropriate.

closed 2018-12-17 by marcoesposito1988