IFL-CAMP/easy_handeye #18: What is meant by Calibrate the robot?

nilsmelchert, 2018-07-30:
Hello, 
in the README.md of this repo it is said, that a robot calibration would lead to better results.
I would really like to calibrate the kinematics of my industrial robot (correct the joint and the robot frame offsets) but haven't found a suitable package for that yet. 

So as it is mentioned here in this repository, I would like to know, how you did your robot calibration in order to get better results?

nilsmelchert, 2018-07-30:
Currently I am aware of two packages which are meant to be for that kind of calibration:
[robot_calibration](http://wiki.ros.org/robot_calibration) and [pr2_calibration](http://wiki.ros.org/pr2_calibration)

Which I don't find very satisfying (see [my question on rosanswers](https://answers.ros.org/question/299042/identification-of-kinematics-of-industrial-robots/)).

marcoesposito1988, 2018-08-07:
Hi @nilsmelchert,

I included this remark from completeness - it was included in the original paper from Tsai, which is now almost 30 years old IIRC.

I guess that if you needed to do a particular calibration procedure, you would be aware about it (or would be at least described in the user manual). The only example I am familiar with is the GMSPositionReferencing application that can be generated through the Sunrise environment of the KUKA iiwa.

Unless you need submillimeter accuracy, you should not need to worry.

nilsmelchert, 2018-08-07:
First of all. Thanks for your reply. I actually do need submillimeter accuracy. I will contribute my package on how to easily calibrate the joint and link offset as soon as I get done with it. 

I will close this Issue now.

closed 2018-08-07 by nilsmelchert

marcoesposito1988, 2018-08-07:
On second thought, I might have misunderstood you: do you mean model identification? I.E., refining the URDF to make the joint length coincide with the physical size? I was referring to calibration of the robot software, rather than ROS side.

In this case, I cannot be of much help either. I refined the joint lengths for iiwa_stack, but it was a brute force method (putting the robot straight, bending the first joint at 90 degrees, taking the z coordinate, and repeating for all successive joints). That made the values on the ROS side coincide with the cabinet's forward kinematics. While this helps a lot having a coherent system, it does not necessarily lead to a better accuracy if the robot software is wrong as well.

nilsmelchert, 2018-08-07:
> Sorry, we wrote at the same time! What robot are you using? If it is the iiwa, you could take a look at this paper. The ideas contained in it could be helpful either way actually, but you might need to make a lot more work.

Thanks a lot. That paper seems to be very helpful.