IFL-CAMP/easy_handeye #20: Reading frames from tf or rviz

GianlucaCerilli, 2018-11-21:
Hello,

I am using this package to calibrate a monocular camera eye_on_hand with a Kinova Jaco2.
I have tried several times but the final result is wrong. I know that the end effector frame in Rviz is not exactly the same as on the real Kinova robot (there is an angle offset), so I wanted to know if this can lead to the wrong result.
I am not sure if the end_effector frame read from this package is the one in Rviz or (uniquely) in what is published in tf.

Thank you very much

marcoesposito1988, 2018-11-25:
That may be a serious problem. It may be worth solving it before going on with anything else.

If the error is systematic, that shouldn't actually compromise the calibration (since the transform between the marker and the robot hand is arbitrary anyway). But if you just want to try out the handeye calibration, in the meantime you can try to fix the marker to a previous link of the robot, which is correctly shown in rviz. If the robot has enough degrees of freedom you should still be able to move the marker in enough directions to perform the calibration successfully. You may have to move the robot by hand instead than through our plugin, though.

GianlucaCerilli, 2018-12-01:
Thank you @marcoesposito1988 , there was a problem in the calibration of the arm and it seems to work.
But honestly the result that I get from aruco is not really accurate (particularly for the Z axis) and so my final extrinsic calibration is not really precise on that axis too.
I mean, usually the Z axis should go from the marker to the sensor of the camera. But in this case it seems to compute the distance from the marker to the lens of the camera. So, there are some centimeters of errors in that direction.
Did you also face the same problem?

Thank you very much

marcoesposito1988, 2018-12-04:
I would first check if the aruco tracking is working correctly. You can do this by leaving the marker on the table, positioning the robot such that the marker is tracked, and moving the robot by hand with its control panel. You can then compare the movement of the robot on the panel with the movement of the marker as reported in tf (when you move the robot 10 cm up, it should move 10 cm in z, and so on...).

If this test fails, you might want to check out the intrinsic calibration of your camera and the marker size with a ruler (it is a dumb mistake, but I wasted many hours because of it).

If the aruco tracking is working, then it may be that you are sampling one transform during calibration, and then publishing the calibration for that frame to another frame at runtime. Before you go into the whole reference_frame/marker_frame story, I would do the check above.

marcoesposito1988, 2019-04-29:
Closing for inactivity. If the problem persists, please comment and the issue will be reopened if appropriate.

closed 2019-04-29 by marcoesposito1988