IFL-CAMP/easy_handeye #21: The program suddenly crashes

WwYyFan, 2018-12-05:
@marcoesposito1988  Hi ,I'm using it with a kinect v2,the program can start up, but it always suddenly crashes,and I don't konw where the problem is.

**Terminal error message**:
```
Traceback (most recent call last):
  File "/home/efun/work/kinect_test/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_calibrationmovements.py", line 251, in handle_next_pose
    self.local_mover.execute_plan(plan)
  File "/home/efun/work/kinect_test/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_calibrationmovements.py", line 110, in execute_plan
    raise RuntimeError("got crazy plan!")
RuntimeError: got crazy plan!
[calibration_mover-6] process has died [pid 29202, exit code -11, cmd /home/efun/work/kinect_test/src/easy_handeye/rqt_easy_handeye/scripts/rqt_calibrationmovements __name:=calibration_mover __log:=/home/efun/.ros/log/328d8954-f875-11e8-8d2c-4cedfbcaaae4/calibration_mover-6.log].
log file: /home/efun/.ros/log/328d8954-f875-11e8-8d2c-4cedfbcaaae4/calibration_mover-6*.log
```

**log file : calibration_mover-6-stdout.log**
```
arguments:  Namespace(quiet=False)
unknowns:  []
^[[0m^[[0m
^[[33m^[[0m
^[[0m^[[0m
^[[0m^[[0m
^[[33m^[[0m
^[[0m^[[0m
^[[33m^[[0m
^[[0m^[[0m
```
**log file : calibration_mover-6.log**
```
[rospy.client][INFO] 2018-12-05 18:04:59,683: init_node, name[/calibration_mover], pid[29202]
[xmlrpc][INFO] 2018-12-05 18:04:59,683: XML-RPC server binding to 127.0.0.1:0
[xmlrpc][INFO] 2018-12-05 18:04:59,683: Started XML-RPC server [http://localhost:44619/]
[rospy.init][INFO] 2018-12-05 18:04:59,684: ROS Slave URI: [http://localhost:44619/]
[rospy.impl.masterslave][INFO] 2018-12-05 18:04:59,684: _ready: http://localhost:44619/
[rospy.registration][INFO] 2018-12-05 18:04:59,684: Registering with master node http://localhost:11311
[xmlrpc][INFO] 2018-12-05 18:04:59,685: xml rpc node: starting XML-RPC server
[rospy.init][INFO] 2018-12-05 18:04:59,784: registered with master
[rospy.rosout][INFO] 2018-12-05 18:04:59,784: initializing /rosout core topic
[rospy.rosout][INFO] 2018-12-05 18:04:59,786: connected to core topic /rosout
[rospy.simtime][INFO] 2018-12-05 18:04:59,786: /use_sim_time is not set, will not subscribe to simulated time [/clock] topic
[rospy.internal][INFO] 2018-12-05 18:05:00,088: topic[/rosout] adding connection to [/rosout], count 0
[rospy.core][INFO] 2018-12-05 18:06:07,418: signal_shutdown [atexit]
[rospy.internal][INFO] 2018-12-05 18:06:07,420: topic[/rosout] removing connection to /rosout
[rospy.impl.masterslave][INFO] 2018-12-05 18:06:07,420: atexit

```

marcoesposito1988, 2018-12-05:
Hi @WwYyFan,

This is a safety measure. The program aborts if the planner finds a trajectory where the robot would move too much (for example, when one joint goes from -90 to 90 degrees). This could result in a big motion and a collision with the environment.

You can adjust the thresholds by passing a list as parameter (I hope it’s somewhere in the documentation, I have to check). But you should then keep an eye on the robot, and a finger on the red button! It is your responsibility to avoid that the robot crashes against something while it moves automatically.

WwYyFan, 2018-12-06:
@marcoesposito1988  thanks a lot,I have changed the robot to another position,and the program is work now.Thanks.

closed 2018-12-06 by WwYyFan