IFL-CAMP/easy_handeye #22: Can't get aruco_maker frame

cygreat7, 2018-12-18:
I run the launch file and look at the tf tree
[frames.pdf](https://github.com/IFL-CAMP/easy_handeye/files/2688527/frames.pdf)
I can't get the aruco_marker frame. so So I tried to run the aruco_ros package separately.

roslaunch   kinect2_bridge   kinect2_bridge.launch
roslaunch   aruco_ros   single.launch
rosrun    image_view  image_view    image:=/aruco_single/result

[image]

I can get the marker information.
[image]

But I still can't get the pose.
[image]

Thank you very much for your advice.

ForsenCD, 2018-12-20:
Hey, 

this worked for me:

1. start you camera launcher

2.  start the handeye launch-file, here is mine:
```
<launch>
    <arg name="namespace_prefix" default="ur5_kinect_handeyecalibration" />
    <arg name="robot_ip" doc="The IP address of the UR5 robot" />
    <arg name="marker_size" value="0.1" doc="Size of the ArUco marker used, in meters" />
    <arg name="marker_id" value="12" doc="The ID of the ArUco marker used" />
    <!-- start the Kinect -->
    <include file="$(find perception)/launch/primesense.launch" >
        <arg name="depth_registration" value="true" />
    </include>
    <node name="aruco_tracker" pkg="aruco_ros" type="single">
        <remap from="/camera_info" to="/primesense/rgb/camera_info" />
        <remap from="/image" to="/primesense/rgb/image_raw" />
        <param name="image_is_rectified" value="true"/>
        <param name="marker_size"        value="0.1"/>
        <param name="marker_id"          value="12"/>
        <param name="reference_frame"    value="primesense_link"/>
        <param name="camera_frame"       value="primesense_rgb_optical_frame"/>
        <param name="marker_frame"       value="camera_marker" />
    </node>
    <!-- start the robot -->
    <include file="$(find ur_modern_driver)/launch/ur5_bringup_joint_limited.launch">
        <arg name="robot_ip" value="192.168.117.117" />
    </include>
    <include file="$(find ur5_moveit_config)/launch/ur5_moveit_planning_execution.launch">
        <arg name="limited" value="true" />
    </include>
    <!-- start easy_handeye -->
    <include file="$(find easy_handeye)/launch/calibrate.launch" >
        <arg name="namespace_prefix" value="$(arg namespace_prefix)" />
        <arg name="eye_on_hand" value="true" />

        <arg name="tracking_base_frame" value="primesense_link" />
        <arg name="tracking_marker_frame" value="camera_marker" />
        <arg name="robot_base_frame" value="base_link" />
        <arg name="robot_effector_frame" value="ee_link" />

        <arg name="freehand_robot_movement" value="true" />
        <arg name="robot_velocity_scaling" value="0.1" />
        <arg name="robot_acceleration_scaling" value="0.1" />
    </include>
</launch>
```

3.  In a new terminal:
`rosrun image_view image_view image:=/aruco_tracker/result`

cygreat7, 2018-12-21:
Thankyou for your reply .I can recognize the marker,but I can't get the marker frame

zhanghengChina, 2019-01-17:
Hello:
I have met the same problem, but now I have solve it. 
First, please modify the parameter in the demo launch file in easy_handeye/docs/example_launch/ur5_kinect_calibration.launch and make sure it is right. The answer above gives another example.
Second, print the aruco in this website http://chev.me/arucogen/ and pls choose Original Aruco. If you choose other aruco, i think this package won't recognize it.
Third, add these to the launch file, it can publish the frame u want.

    <node pkg="aruco_ros" type="marker_publisher" name="aruco_marker_publisher">
        <remap from="/camera_info" to="/cameras/rgb/camera_info" />
        <remap from="/image" to="/camera/rgb/image_rect_color" />
        <param name="image_is_rectified" value="True"/>
        <param name="marker_size"        value="$(arg marker_size)"/>
        <param name="reference_frame"    value="camera_link"/>   
        <param name="camera_frame"       value="$camera_rgb_optical_frame"/>
    </node>

It is the final launch file i used.
<launch>
    <arg name="namespace_prefix" default="ur5_realsense_handeyecalibration" />

    <arg name="marker_size" value="0.1"/>
    <arg name="marker_id" value="50"/>

    
    <include file="$(find realsense_camera)/launch/sr300_nodelet_rgbd.launch" >
    </include>

    
    <node name="aruco_tracker" pkg="aruco_ros" type="single">
        <remap from="/camera_info" to="/camera/rgb/camera_info" />
        <remap from="/image" to="/camera/rgb/image_rect_color" />
        <param name="image_is_rectified" value="true"/>
        <param name="marker_size"        value="$(arg marker_size)"/>
        <param name="marker_id"          value="$(arg marker_id)"/>
        <param name="reference_frame"    value="camera_link"/>
        <param name="camera_frame"       value="camera_rgb_optical_frame"/>
        <param name="marker_frame"       value="camera_marker" />
    </node>

    <node pkg="aruco_ros" type="marker_publisher" name="aruco_marker_publisher">
        <remap from="/camera_info" to="/cameras/rgb/camera_info" />
        <remap from="/image" to="/camera/rgb/image_rect_color" />
        <param name="image_is_rectified" value="True"/>
        <param name="marker_size"        value="$(arg marker_size)"/>
        <param name="reference_frame"    value="camera_link"/>   
        <param name="camera_frame"       value="$camera_rgb_optical_frame"/>
    </node>

  <include file="$(find ur5_table_moveit_202)/launch/ur5_planning_execution.launch"/>

    
    <include file="$(find easy_handeye)/launch/calibrate.launch" >
        <arg name="namespace_prefix" value="$(arg namespace_prefix)" />
        <arg name="eye_on_hand" value="true" />

        <arg name="tracking_base_frame" value="camera_link" />
        <arg name="tracking_marker_frame" value="camera_marker" />
        <arg name="robot_base_frame" value="base" />
        <arg name="robot_effector_frame" value="tool0" />

        <arg name="freehand_robot_movement" value="false" />
        <arg name="robot_velocity_scaling" value="0.5" />
        <arg name="robot_acceleration_scaling" value="0.2" />
    </include>

</launch>

marcoesposito1988, 2019-04-29:
Closing for inactivity. If the problem persists, please comment and the issue will be reopened if appropriate.

closed 2019-04-29 by marcoesposito1988

lucamarchionna, 2021-07-05:
Hi everyone, I was trying to a simple work based on aruco. However, I don't find the package aruco_ros package for Noetic. In particular, the file I am searching for is aruco_ros/src/single, which is the same file you used.
 I have tried to search on GitHub but the similar packages seem a little bit different. Can anyone indicate the right repository?

marcoesposito1988, 2021-07-07:
Hi @lucamarchionna,

this is the original repository for aruco_ros: https://github.com/pal-robotics/aruco_ros

However I was so frustrated that I wrote my own wrapper for the OpenCV aruco tracking, maybe it can be helpful for you too: https://github.com/marcoesposito1988/easy_aruco