IFL-CAMP/easy_handeye #23: Transformation seems to be wrong for eye-on-hand

ForsenCD, 2018-12-20:
Hi,

first of all thank you for this tool. It was easy to setup and it looks to be promising for my project.

I used the gui to get the **eye-on-hand** transformation for my Asus Xtion and the UR5. The camera is mounted on the end-effector. I am trying to transform camera coordinates from a tracked object to the end effector frame of my UR5 and the move the robot to those coordinates. But somehow the transformation does not give me the expected results. Here's my data:

### Transformation.yaml
```

eye_on_hand: true
robot_effector_frame: ee_link
tracking_base_frame: primesense_link
transformation: {qw: 0.12315000298790331, qx: -0.989347167929719, qy: -0.07459459276842946,
qz: 0.02149197066897385, x: 0.04013691589594168, y: -0.13677498325817272, z: 0.08220400023605642}
```

### Grasp_Camera_data
```
   position: 
       x: -0.0987664492927
       y: -0.0910226523694
       z: 0.560733250459
   orientation: 
       x: 0.631932686942
       y: 0.31727130216
       z: 0.631932686942
       w: -0.31727130216
  q_value: 0.678789377213
  grasp_type: 0
  center_px: [226.52753499783466, 153.77785122466588]
  angle: -0.640199793927
  depth: 0.560733250459
  ...
```

### Camera_ee_transform:
```
header: 
seq: 0
frame_id: "primesense_link"
child_frame_id: "ee_link"
transform: 
translation: 
    x: 0.040137
    y: -0.136775
    z: 0.082204
rotation: 
    x: -0.989347006868672
    y: -0.0745944036699313
    z: 0.021491953634446583
    w: 0.12315016789268264
```
Then I used t2f to calcualte the grasp in the ee pose
`
grasp_ee_pose = tf2_geometry_msgs.do_transform_pose(grasp_camera_pose, camera_ee_transform)`

The result is incorrect. Here are the actual results and my expected results for the translation:
```
grasp_ee_pose_translation_actual:
x = -0.104539997152
y  = 0.0702126802005
z = -0.436145472381
```
```

grasp_ee_pose_translation_expected:

x =  0.362407802611
y =  0.667193262606
z =  0.050243921518
```
What am I missing? Thank you!

marcoesposito1988, 2018-12-21:
Does the xtion driver publish tf transformations on his own? In that case, you should have the same exact problem as [here](https://github.com/IFL-CAMP/easy_handeye/issues/19#issuecomment-449356180)

ForsenCD, 2018-12-21:
> Does the xtion driver publish tf transformations on his own? In that case, you should have the same exact problem as [here](https://github.com/IFL-CAMP/easy_handeye/issues/19#issuecomment-449356180)

Hey! Thank you for the answer :) I looked at the other issue but I don't think that I have the same issue since I used the "primesense_link" as my reference frame. Here's my launch file:

```
<launch>
    <arg name="namespace_prefix" default="ur5_kinect_handeyecalibration" />

    <arg name="robot_ip" doc="The IP address of the UR5 robot" />

    <arg name="marker_size" value="0.1" doc="Size of the ArUco marker used, in meters" />
    <arg name="marker_id" value="12" doc="The ID of the ArUco marker used" />

    <!-- start the Kinect -->
    <include file="$(find perception)/launch/primesense.launch" >
    </include>

    <node name="aruco_tracker" pkg="aruco_ros" type="single">
        <remap from="/camera_info" to="/primesense/rgb/camera_info" />
        <remap from="/image" to="/primesense/rgb/image_raw" />
        <param name="image_is_rectified" value="true"/>
        <param name="marker_size"        value="0.1"/>
        <param name="marker_id"          value="12"/>
        <param name="reference_frame"    value="primesense_link"/>
        <param name="camera_frame"       value="primesense_rgb_optical_frame"/>
        <param name="marker_frame"       value="camera_marker" />
    </node>

    <!-- start the robot -->
    <include file="$(find ur_modern_driver)/launch/ur5_bringup_joint_limited.launch">
        <arg name="robot_ip" value="192.168.117.117" />
    </include>
    <include file="$(find ur5_moveit_config)/launch/ur5_moveit_planning_execution.launch">
        <arg name="limited" value="true" />
    </include>

    <!-- start easy_handeye -->
    <include file="$(find easy_handeye)/launch/calibrate.launch" >
        <arg name="namespace_prefix" value="$(arg namespace_prefix)" />
        <arg name="eye_on_hand" value="true" />

        <arg name="tracking_base_frame" value="primesense_link" />
        <arg name="tracking_marker_frame" value="camera_marker" />
        <arg name="robot_base_frame" value="base_link" />
        <arg name="robot_effector_frame" value="ee_link" />

        <arg name="freehand_robot_movement" value="false" />
        <arg name="robot_velocity_scaling" value="0.1" />
        <arg name="robot_acceleration_scaling" value="0.1" />
    </include>

</launch>
```

I get some weird messages though but I don't know if it's related to my problem:
`[WARN] [1545415568.476705]: Detected jump back in time of 0.000494s. Clearing TF buffer.`

And here is my setup:

[image]

marcoesposito1988, 2018-12-22:
Yes, that's right.. sorry, I overlooked that.

Are you using the easy_handeye publisher to publish the calibration? I might be wrong again, but it doesn't look like that: in your camera_ee_transform you are publishing from the camera to the robot hand, usually i always had it in the opposite direction. I also don't understand how publishing from the camera to the robot hand may play well with the ur5 robot state publisher, since it should already publish the whole tf frames of the robot including ee_link. If both the robot state publisher and you publish a transform with ee_link as child frame, tf should insult you heavily. 

If I remember correctly, the result you get for this configuration is the transform from the hand to the camera (the easy_handeye publisher takes care of publishing it in the correct direction). If you have the result you get in the opposite direction, that would be wrong.

If you are publishing it by hand instead of with our publisher, you might fix it by inverting the transform (since it looks like it is in the wrong direction). If this is the case.. merry christmas! You just avoided some hours of debugging :stuck_out_tongue_closed_eyes:

ForsenCD, 2018-12-22:
Hey marcoesposito1 thanks for the feedback!
Yes I think you are right, I used the wrong transformation. But I already tried the inverted transformation but it was also wrong. I think it's not even the transformation that's wrong (because it looks like it's correct). The center of the camera is about 13 cm away from the ee and it's rotated 180° so it might be correct. 

I think the problem is using the `moveit_commander.PlanningSceneInterface()` with the transformation since `moveit_commander.MoveGroupCommander('manipulator').set_pose_target(pose_goal)`
uses the reference frame `base_link` to  plan a motion  to a desired pose for the end-effector.
I think using another reference frame here might fix the problem. I will try it in a few days. Until then have a nice holidays :smile:

marcoesposito1988, 2019-04-29:
Closing for inactivity. If the problem persists, please comment and the issue will be reopened if appropriate.

closed 2019-04-29 by marcoesposito1988

EmilyJrxx, 2019-08-24:
Hi! I am a beginner and i might encounter with a similar problem. My realsense D435 camera is fixed rigidly on the end of the ur5, as is shown below
[image]
and here is my ur5_realsense_handeyecalibration.launch file:
`  <launch>
        <arg name="namespace_prefix" default="ur5_realsense_handeyecalibration" />
     
        <arg name="robot_ip" doc="The IP address of the UR5 robot" />
    		
        <arg name="marker_size" doc="Size of the ArUco marker used, in meters" default="0.079" />
        <arg name="marker_id" doc="The ID of the ArUco marker used" default="582" />
     
        
        
       
        
            
        
        
     
        
        <node name="aruco_tracker" pkg="aruco_ros" type="single">
            <remap from="/camera_info" to="/camera/color/camera_info" />
            <remap from="/image" to="/camera/color/image_raw" />
            <param name="image_is_rectified" value="true"/>
            <param name="marker_size"        value="$(arg marker_size)"/>
            <param name="marker_id"          value="$(arg marker_id)"/>
            <param name="reference_frame"    value="camera_link"/>
            <param name="camera_frame"       value="camera_color_optical_frame"/>          
            <param name="marker_frame"       value="camera_marker" />
        </node>
     
  
    
    
        <include file="$(find easy_handeye)/launch/calibrate.launch" >
            <arg name="eye_on_hand" value="true" />
            <arg name="namespace_prefix" value="$(arg namespace_prefix)" />
            
              
            
            
            <arg name="robot_base_frame" value="base_link" />
            
            <arg name="robot_effector_frame" value="ee_link" />
            
            <arg name="tracking_base_frame" value="camera_link" />
            <arg name="tracking_marker_frame" value="camera_marker" />
     
            <arg name="freehand_robot_movement" value="false" />
            <arg name="robot_velocity_scaling" value="0.5" />
            <arg name="robot_acceleration_scaling" value="0.2" />
        </include>
     
    </launch>`
I successfully runned the calibration procedure and get 15 samples out of 17 postions, no error occured but the result looked rather weird shown in rviz, i use a 'tf' node to publish the result and display it in rivz

`<node pkg="tf" type="static_transform_publisher" name="ur5_broadcaster" args="-0.05492695051566381 0.08851516098866889 0.04179254513490418 -0.2245430153143076-0.24578677988435596 0.6655710116549832 0.6679704496100108 base_link camera_link 100" />`

[image]
Can you give me any advice? thank you!

marcoesposito1988, 2019-08-26:
Hello @EmilyJrxx, 

you are performing an eye-in-hand calibration which gives you the transformation between the end effector and the camera. However, after calibration, you are publishing this transform from the base (`base_link`) instead of from the end effector (`ee_link`):
`<node pkg="tf" type="static_transform_publisher" name="ur5_broadcaster" args="... [!!!!ERROR IS HERE!!!!:]base_link camera_link 100" />`

Actually, you don't need to do this yourself! easy_handeye was written precisely to avoid such error-prone processes of copy-pasting. In your launch file, instead of this static_transform_publisher, you can include the easy_handeye publisher:
```xml
<launch>
  <include file="$(find easy_handeye)/launch/publish.launch">
    <arg name="namespace" value="YOUR_NAMESPACE"/>
  </include>
</launch>
```
All you have to adjust is the value of the namespace argument, which should coincide with the value of that argument during calibration. If you calibrate again, the new value will be automatically used.

EmilyJrxx, 2019-08-28:
Thank you for replying so soon! I found the problem you mentioned right before i checked your reply by  reading the src code handeye_calibration.py! I've done another few tests and adjustments with teamates in the lab, now the position (translation) of camera_link is corrected but the orientation is still wrong,
[image]
(it looks that the orientation of camera_link is not random, and i can fix it by several 90-degree-rotation)

so  i took a  'brutal' way to deal with this by multiplying the rotation quaternion given by the calibration with another quaternion to rotate it, and now the orientation of camera_link looks right.

and now it looks right, i kind of fixed this problem though i still can not tell what's wrong in my procedure. Thanks for your help!

marcoesposito1988, 2019-08-29:
Probably it is related to the fact that you need to use the `reference_frame` argument, as shown in `docs/example_launch/ur5_kinect_calibration.launch`.

This is necessary because RGBD cameras are represented not through a single tf reference frame, but through several ones. While ArUco provides the transform between the optical center of the camera and the marker, we need the transform between the root of the camera tf tree (the reference frame) and the marker.