IFL-CAMP/easy_handeye #25: question about the results of eye on base calibration

hahazhky, 2019-01-19:
Thanks for your code.
when I use this to do my eye on base calibration, the results show 

Computed calibration: effector_camera:

It seems like the eye in hand calibration results, what should I do to get the base_camere matrix？

marcoesposito1988, 2019-01-23:
Hi @hahazhky,

the matrix is printed for debugging/informative purposes only. It is actually the message returned by the ViSP ROS wrapper, which uses that identifier regardless of the type of calibration (eye on hand or eye on base).

If you want to perform the eye on base calibration, you should set the roslaunch parameter eye_on_hand to false. You can ignore that identifier.

marcoesposito1988, 2019-04-29:
Closing for inactivity. If the problem persists, please comment and the issue will be reopened if appropriate.

closed 2019-04-29 by marcoesposito1988