IFL-CAMP/easy_handeye #26: [ERROR] [1548085881.240854]: Error processing request: canTransform: source_frame marker does not exist.

1gen100, 2019-01-21:
[ERROR] [1548085881.240854]: Error processing request: canTransform: source_frame marker does not exist.
['Traceback (most recent call last):\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 625, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/home/student/catkin_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_server.py", line 38, in take_sample\n    self.calibrator.take_sample()\n', '  File "/home/student/catkin_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_calibrator.py", line 140, in take_sample\n    transforms = self._get_transforms()\n', '  File "/home/student/catkin_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_calibrator.py", line 123, in _get_transforms\n    time = self._wait_for_transforms()\n', '  File "/home/student/catkin_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_calibrator.py", line 111, in _wait_for_transforms\n    self.listener.waitForTransform(self.tracking_base_frame, self.tracking_marker_frame, now, rospy.Duration(10))\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf/listener.py", line 76, in waitForTransform\n    raise tf2_ros.TransformException(error_msg or "no such transformation: \\"%s\\" -> \\"%s\\"" % (source_frame, target_frame))\n', 'TransformException: canTransform: source_frame marker does not exist.\n']
Traceback (most recent call last):
  File "/home/student/catkin_ws/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_easy_handeye.py", line 104, in handle_take_sample
    sample_list = self.client.take_sample()
  File "/home/student/catkin_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_client.py", line 34, in take_sample
    return self.take_sample_proxy().samples
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 515, in call
    responses = transport.receive_once()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 727, in receive_once
    p.read_messages(b, msg_queue, sock) 
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 353, in read_messages
    self._read_ok_byte(b, sock)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 336, in _read_ok_byte
    raise ServiceException("service [%s] responded with an error: %s"%(self.resolved_name, str))
rospy.service.ServiceException: service [/iiwa_kinect_calibration_eye_hand_eye_on_base/take_sample] responded with an error: error processing request: canTransform: source_frame marker does not exist.
[iiwa_kinect_calibration_eye_hand_eye_on_base/namespace_student_ThinkPad_W541_7532_6188026501665740049_rqt-17] process has died [pid 7756, exit code 1, cmd /opt/ros/kinetic/lib/rqt_gui/rqt_gui --clear-config --perspective-file /home/student/catkin_ws/src/easy_handeye/easy_handeye/launch/rqt_easy_handeye.perspective __name:=namespace_student_ThinkPad_W541_7532_6188026501665740049_rqt __log:=/home/student/.ros/log/41fd7db0-1d94-11e9-8362-a4c494410bb3/iiwa_kinect_calibration_eye_hand_eye_on_base-namespace_student_ThinkPad_W541_7532_6188026501665740049_rqt-17.log].
log file: /home/student/.ros/log/41fd7db0-1d94-11e9-8362-a4c494410bb3/iiwa_kinect_calibration_eye_hand_eye_on_base-namespace_student_ThinkPad_W541_7532_6188026501665740049_rqt-17*.log

1gen100, 2019-01-21:
What's the problem I can't find out how to make it work, help???

marcoesposito1988, 2019-01-23:
The problem seems to be that the tf frame "marker" is not being published when you sample the tranforms.

Could you check?

1gen100, 2019-01-23:
Thanks, the problem was that my `rqt_gui` doesn't work for me have tried to reinstall and all that but it want wotk so I tried to run `rqt_image_view` instead and it worked.

But another problem I have is that the rviz for the handeye dosn't get the robot descreption and because of thatb the motion planing doesn't work on the rviz for handeye only on the moveit.rviz, it gave me this ERRORS.
[ERROR] [1548241712.696638044]: Robot model parameter not found! Did you remap 'robot_description'?
[ERROR] [1548241712.698695320]: Robot model not loaded

marcoesposito1988, 2019-06-17:
Closing for inactivity. If the problem persists, please comment and the issue will be reopened if appropriate.

closed 2019-06-17 by marcoesposito1988

1gen100, 2019-06-19:
Hi again, I was trying to use the automatic movement function, but i won't work because of the error about the robot model:

[ERROR] [1560934411.067585300]: Robot model parameter not found! Did you remap 'robot_description'?
[ERROR] [1560934411.069075347]: Robot model not loaded
Ass you can see on this picture the ghost of the robot dosn't appear, Do you have any solution for this problem?
[image]

1gen100, 2019-06-19:
Error when i try runing the launch file with for the auto movement:
`<arg name="freehand_robot_movement" value="false" />`

[ERROR] [1560934812.091702728]: Robot model parameter not found! Did you remap 'robot_description'?
[FATAL] [1560934812.091753445]: Unable to construct robot model. Please make sure all needed information is on the parameter server.
PluginManager._load_plugin() could not load plugin "rqt_easy_handeye/Hand-eye Calibration automatic movement":
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler.py", line 99, in load
    self._load()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler_direct.py", line 54, in _load
    self._plugin = self._plugin_provider.load(self._instance_id.plugin_id, self._context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui_py/ros_py_plugin_provider.py", line 60, in load
    return super(RosPyPluginProvider, self).load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui/ros_plugin_provider.py", line 101, in load
    return class_ref(plugin_context)
  File "/home/dmri/catkin_ws/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_calibrationmovements.py", line 296, in __init__
    self._widget = CalibrationMovementsGUI()
  File "/home/dmri/catkin_ws/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_calibrationmovements.py", line 149, in __init__
    self.local_mover = CalibrationMovements(move_group_name, max_velocity_scaling, max_acceleration_scaling)
  File "/home/dmri/catkin_ws/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_calibrationmovements.py", line 26, in __init__
    self.mgc = MoveGroupCommander(move_group_name)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/moveit_commander/move_group.py", line 52, in __init__
    self._g = _moveit_move_group_interface.MoveGroupInterface(name, robot_description, ns)
RuntimeError: Unable to construct robot model. Please make sure all needed information is on the parameter server.

[calibration_mover-19] process has died [pid 23405, exit code 1, cmd /home/dmri/catkin_ws/src/easy_handeye/rqt_easy_handeye/scripts/rqt_calibrationmovements __name:=calibration_mover __log:=/home/dmri/.ros/log/a1714a82-9270-11e9-a70c-54ee75886762/calibration_mover-19.log].
log file: /home/dmri/.ros/log/a1714a82-9270-11e9-a70c-54ee75886762/calibration_mover-19*.log