IFL-CAMP/easy_handeye #27: I can't load plugin "rqt_easy_handeye/Hand-eye Calibration automatic movement"

iamyk666, 2019-01-23:
PluginManager._load_plugin() could not load plugin "rqt_easy_handeye/Hand-eye Calibration automatic movement":
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler.py", line 99, in load
    self._load()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler_direct.py", line 54, in _load
    self._plugin = self._plugin_provider.load(self._instance_id.plugin_id, self._context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui_py/ros_py_plugin_provider.py", line 60, in load
    return super(RosPyPluginProvider, self).load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui/ros_plugin_provider.py", line 101, in load
    return class_ref(plugin_context)
  File "/home/irec/catkin_ws/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_calibrationmovements.py", line 296, in __init__
    self._widget = CalibrationMovementsGUI()
  File "/home/irec/catkin_ws/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_calibrationmovements.py", line 149, in __init__
    self.local_mover = CalibrationMovements(move_group_name, max_velocity_scaling, max_acceleration_scaling)
  File "/home/irec/catkin_ws/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_calibrationmovements.py", line 26, in __init__
    self.mgc = MoveGroupCommander(move_group_name)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/moveit_commander/move_group.py", line 52, in __init__
    self._g = _moveit_move_group_interface.MoveGroupInterface(name, robot_description, ns)
RuntimeError: Unable to connect to move_group action server 'move_group' within allotted time (5s)

[calibration_mover-35] process has died [pid 6368, exit code 1, cmd /home/irec/catkin_ws/src/easy_handeye/rqt_easy_handeye/scripts/rqt_calibrationmovements __name:=calibration_mover __log:=/home/irec/.ros/log/1aea8c3a-1edf-11e9-aaf0-7085c23896f7/calibration_mover-35.log].
log file: /home/irec/.ros/log/1aea8c3a-1edf-11e9-aaf0-7085c23896f7/calibration_mover-35*.log
[ WARN] [1548228569.887509930, 629.885000000]: Waiting for /follow_joint_trajectory to come up
[ERROR] [1548228575.893280482, 635.886000000]: Action client not connected: /follow_joint_trajectory

marcoesposito1988, 2019-01-23:
Have you run catkin_make and sourced the workspace?

iamyk666, 2019-01-23:
I can't get the GUI of collecting pictures.

saratrajput, 2019-03-18:
Hello, I have the same problem. Did you find a solution?

iamyk666, 2019-03-18:
> Hello, I have the same problem. Did you find a solution?

Yes,I have done the calibration of the camera. You can try run the robot first, and then roslaunch the file of calibration like this:
$ roslaunch ur_modern_driver ur5_bringup.launch robot_ip:=192.168.1.10 use_lowbandwidth_trajectory_follower:=true
$ roslaunch easy_handeye ur5_kinect_calibration.launch

saratrajput, 2019-03-20:
I already have the TM robot and the camera running on a separate master. I'm just subscribing to them. I think it has something to do with Ubuntu 18.04 and ROS Melodic perhaps. I'll try docker.

saratrajput, 2019-03-20:
I still have the same problem inside Docker. Can you tell me how you loaded the plugin "rqt_easy_handeye/Hand-eye Calibration automatic movement"?

aPonza, 2019-07-10:
see [this](https://github.com/IFL-CAMP/easy_handeye/issues/1#issuecomment-509984232)?

marcoesposito1988, 2020-03-14:
Should be a duplicate of #1

closed 2020-03-14 by marcoesposito1988