IFL-CAMP/easy_handeye #28: Hellow,I run easy_handeye with gazebo (a launch file like you another file named easy_hand_demo),and I have some problem as follow.Please help me solve it , and thank you provide this useful demo!

iamyk666, 2019-01-25:
FIRST,THIS IS MY LAUNCH FILE.(You can copy this launch file to your /easy_hand/src/launch/  and run it as my steps as follow to copy this problem,thank you. )
__________________________________________________________________________
<launch>
    <arg name="namespace_prefix" default="ur5_kinect_handeyecalibration" />
    <arg name="robot_ip" doc="The IP address of the UR5 robot" />
    <arg name="marker_size" doc="Size of the ArUco marker used, in meters" />
    <arg name="marker_id" doc="The ID of the ArUco marker used" />

    
    <include file="$(find freenect_launch)/launch/freenect.launch" >
        <arg name="depth_registration" value="true" />
    </include>  

    
    <node name="aruco_tracker" pkg="aruco_ros" type="single">
        <remap from="/camera_info" to="/camera/rgb/camera_info" />
        <remap from="/image" to="/camera/rgb/image_raw" />
        <param name="image_is_rectified" value="true"/>
        <param name="marker_size"        value="0.096"/>
        <param name="marker_id"          value="528"/>
        <param name="reference_frame"    value="camera_link"/>
        <param name="camera_frame"       value="camera_rgb_optical_frame"/>
        <param name="marker_frame"       value="camera_marker" />
    </node>

    
    <include file="$(find ur5_moveit_config)/launch/demo.launch">
        <arg name="limited" value="true" />
    </include>

    
    <include file="$(find easy_handeye)/launch/calibrate.launch" >
        <arg name="namespace_prefix" value="$(arg namespace_prefix)" />
        <arg name="eye_on_hand" value="false" />
        <arg name="tracking_base_frame" value="camera_link" />
        <arg name="tracking_marker_frame" value="camera_marker" />
        <arg name="robot_base_frame" value="base_link" />
        <arg name="robot_effector_frame" value="wrist_3_link" />

        <arg name="freehand_robot_movement" value="false" />
        <arg name="robot_velocity_scaling" value="0.5" />
        <arg name="robot_acceleration_scaling" value="0.2" />
    </include>

</launch>
___________________________________________________________________________________

I want use it to do the easy_hand calibration with gazebo,And I used kinect1.0 camera. The steps are as follow:
First I run command   $ roslaunch freenect_launch freenect.launch   to  start the camera.
Next , I run command  $ roslaunch ur_gazebo to establish a simulate environment.
Finally I run command $ roslaunch easy_handeye ur5_kinect_calibration.launch to do easy_hand calibration.

There are some warning in the $ roslaunch easy_handeye ur5_kinect_calibration.launch window as follow:**
___________________________________________________________________________________
[ WARN] [1548377886.673253632, 21.800000000]: Could not find any compatible image output mode for 1. Falling back to default image output mode 1.
[ WARN] [1548377886.673301001, 21.800000000]: Could not find any compatible depth output mode for 1. Falling back to default depth output mode 1.
[ WARN] [1548377886.687876734, 21.814000000]: Camera calibration file /home/irec/.ros/camera_info/rgb_A00365908893111A.yaml not found.
[ WARN] [1548377886.687926797, 21.814000000]: Using default parameters for RGB camera calibration.
[ WARN] [1548377886.687969401, 21.814000000]: Camera calibration file /home/irec/.ros/camera_info/depth_A00365908893111A.yaml not found.
[ WARN] [1548377886.688220450, 21.815000000]: Using default parameters for IR camera calibration.
[ WARN] [1548377887.219417598, 22.341000000]: Skipping virtual joint 'fixed_base' because its child frame 
[ WARN] [1548377888.323194463, 23.447000000]: 'base_link' does not match the URDF frame 'world'.
Deprecation warning: Trajectory execution service is deprecated (was replaced by an action).
Replace 'MoveGroupExecuteService' with 'MoveGroupExecuteTrajectoryAction' in move_group.launch.
[ WARN] [1548377888.436476686, 23.558000000]: Interactive marker 'EE:goal_ee_link' contains unnormalized quaternions. This warning will only be output once but may be true for others; enable DEBUG messages for ros.rviz.quaternions to see more details.
______________________________________________________________________________________
And when the (0/17) GUI quited immediately there was a error in the $ roslaunch easy_handeye ur5_kinect_calibration.launch window as follow:
____________________________________________________________________________________
Traceback (most recent call last):
  File "/home/irec/catkin_ws/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_calibrationmovements.py", line 251, in handle_next_pose
    self.local_mover.execute_plan(plan)
  File "/home/irec/catkin_ws/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_calibrationmovements.py", line 110, in execute_plan
    raise RuntimeError("got crazy plan!")
RuntimeError: got crazy plan!
[calibration_mover-35] process has died [pid 15198, exit code 1, cmd /home/irec/catkin_ws/src/easy_handeye/rqt_easy_handeye/scripts/rqt_calibrationmovements __name:=calibration_mover __log:=/home/irec/.ros/log/e2fa2cb4-203b-11e9-aaf0-7085c23896f7/calibration_mover-35.log].
log file: /home/irec/.ros/log/e2fa2cb4-203b-11e9-aaf0-7085c23896f7/calibration_mover-35*.log
_____________________________________________________________________________________
And I couldn't take sample at the same time . when I click "TAKE SAMPLE" , it will be no response ,then quit. 

And these are my problems , please help me solve it, I can't find any example like it on the website.

iamyk666, 2019-01-25:
When the "TAKE SAMPLE" is no response ,and quit,there are some errors as follow:
_______________________________________________________________________________________\
[ERROR] [1548381003.191268, 263.534000]: Error processing request: canTransform: source_frame camera_marker does not exist.
['Traceback (most recent call last):\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 625, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/home/irec/catkin_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_server.py", line 38, in take_sample\n    self.calibrator.take_sample()\n', '  File "/home/irec/catkin_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_calibrator.py", line 140, in take_sample\n    transforms = self._get_transforms()\n', '  File "/home/irec/catkin_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_calibrator.py", line 123, in _get_transforms\n    time = self._wait_for_transforms()\n', '  File "/home/irec/catkin_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_calibrator.py", line 111, in _wait_for_transforms\n    self.listener.waitForTransform(self.tracking_base_frame, self.tracking_marker_frame, now, rospy.Duration(10))\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf/listener.py", line 76, in waitForTransform\n    raise tf2_ros.TransformException(error_msg or "no such transformation: \\"%s\\" -> \\"%s\\"" % (source_frame, target_frame))\n', 'TransformException: canTransform: source_frame camera_marker does not exist.\n']
Traceback (most recent call last):
  File "/home/irec/catkin_ws/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_easy_handeye.py", line 104, in handle_take_sample
    sample_list = self.client.take_sample()
  File "/home/irec/catkin_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_client.py", line 34, in take_sample
    return self.take_sample_proxy().samples
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 515, in call
    responses = transport.receive_once()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 727, in receive_once
    p.read_messages(b, msg_queue, sock) 
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 353, in read_messages
    self._read_ok_byte(b, sock)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 336, in _read_ok_byte
    raise ServiceException("service [%s] responded with an error: %s"%(self.resolved_name, str))
rospy.service.ServiceException: service [/ur5_kinect_handeyecalibration_eye_on_base/take_sample] responded with an error: error processing request: canTransform: source_frame camera_marker does not exist.
[ur5_kinect_handeyecalibration_eye_on_base/namespace_irec_desktop_21850_5975872909915625587_rqt-10] process has died [pid 21920, exit code 1, cmd /opt/ros/kinetic/lib/rqt_gui/rqt_gui --clear-config --perspective-file /home/irec/catkin_ws/src/easy_handeye/easy_handeye/launch/rqt_easy_handeye.perspective __name:=namespace_irec_desktop_21850_5975872909915625587_rqt __log:=/home/irec/.ros/log/e24ff9e0-2042-11e9-aaf0-7085c23896f7/ur5_kinect_handeyecalibration_eye_on_base-namespace_irec_desktop_21850_5975872909915625587_rqt-10.log].
log file: /home/irec/.ros/log/e24ff9e0-2042-11e9-aaf0-7085c23896f7/ur5_kinect_handeyecalibration_eye_on_base-namespace_irec_desktop_21850_5975872909915625587_rqt-10*.log

SamNew1, 2019-04-10:
Hi mate, have you solved it? I have the similar problem now, I think we need to define target_frame and source_frame, right?

marcoesposito1988, 2019-04-10:
This error: 
```
raise RuntimeError("got crazy plan!")
RuntimeError: got crazy plan!
```
means that it was not possible to find a motion plan for the robot. This can happen if the starting position is bad (the robot should be able to rotate the end effector about all axis for at least 25 degrees, and translate it 10 cm in all directions).

This other error:
```
canTransform: source_frame camera_marker does not exist.
```
should be self explaining. Did you somehow simulate a tracking system in gazebo?

marcoesposito1988, 2020-03-14:
Closing for lack of feedback

closed 2020-03-14 by marcoesposito1988