IFL-CAMP/easy_handeye #30: TF extrapolation into the past error - Solved

JRauer, 2019-03-05:
Hi,
thank you for your great code! I didn't got it working with your aruco_ros implementation somehow and used aruco_detect with a tf-broadcaster instead. I encountered the problem, that tf throws an "extrapolation into the past"-error because of timing issues. It could be solved by calling self._get_transforms() in handeye_calibrator.py (line 141) with the argument rospy.Time(0). I searched the web and found this: https://answers.ros.org/question/243179/whats-the-diff-between-rospytime0-and-rospytimenow/
saying "cause lookupTransform(..) and friends to return the **latest available data** for a specific transform, instead of the **data at a specific point in time**."
Maybe it would be good to change that in future versions?

marcoesposito1988, 2019-03-10:
Hi @JRauer,

that would work, but it is rather a hack that could mask real problems with your setup. It is like putting black tape on the red "engine failure" light.

Using Time(0), you tell tf to just give you the last available value for that transformation, no questions asked. tf will be happy to do that, but it might give you garbage as result. For example, if the robot goes outside of the field of view of the tracking system, the tf frame might remain stuck at its border. If you use Time(0), tf might give you that (incorrect) value. The result is that you would be happy at first that you managed to finish the calibration, only to bang your head to the wall for a week because the result of the calibration is incorrect (or come and create an issue here because the library "doesn't work").

There are edge cases possible where this could help. For example if you have multiple nodes on a network, and their clocks are way off. However, this is a problematic setup that is going to create headaches (at least) with tf later anyway, and the problem may be solved easily with NTP synchronization (e.g. with chrony).

In light of these considerations, I would rather be against this change. If any compelling reason pops up, I would consider to add the option to force this, e.g. as an optional launch argument with an ugly name. 

Thanks anyway for the suggestion!

JRauer, 2019-03-14:
@marcoesposito1988 Thank you for your reply. I totally understand what's the Problem with this solution and am aware that I have to be careful that I don't record old tf-data when the marker is not visible. It although may be a solution for developers with a tf-problems like me but I am totally with you - this change should only be made by users aware of the consequences.

closed 2019-03-14 by JRauer