IFL-CAMP/easy_handeye #33: 'module' object has no attribute 'TransformListener'

LiGengLei, 2019-03-29:
hello, when I use the command "roslaunch easy_handeye calibrate.launch" to bring up the server, I meet an error as followed:
"AttributeError: 'module' object has no attribute 'TransformListener'"
Can you give me some advice? Thank you very much. I have installed the tf and tf2 by apt-get. My system is Ubuntu.

LiGengLei, 2019-03-31:
I find it is the question of tf. When I instead of using import tf2_ros, then everything right.

marcoesposito1988, 2019-04-01:
Hi @LiGengLei,

tf2_ros was first introduced with pull request #31. However, that only modified the publisher, so I don't understand how this could happen to you during calibration.

Where did you have to change from tf2_ros to tf? Can you please give me the file name and line number?

LiGengLei, 2019-04-02:
@marcoesposito1988 
I am very happy to hear from you. when I meet this question, I search online, and find someone say that tf2_ros package is better than tf. You can read this link: https://github.com/ros/geometry/issues/117. So, I change the source code as follows:
In line 52, I add "tfBuffer=tf2_ros.Buffer(); self.listener = tf2_ros.TransformListener(tfBuffer)"
In line 58, I change to "self.broadcaster = tf2_ros.TransformBroadcaster()"
And because I don't know how to change the line 64 using tf2_ros, so I comment out it.
That's all my change. Maybe my situation is special, because most people use this package don't have this question. I think maybe your code is no question, it's question of my computer environment. 
Lastly, I want to thank you for your free and beautiful code.

marcoesposito1988, 2019-04-02:
Yes, I also want to switch to tf2 but the API is different, so the changes may have to be bigger than that.

You don't have to worry for now, the first version of tf is still supported. You can run my code without changes, and it should work.

marcoesposito1988, 2019-06-17:
Closing for inactivity. If the problem persists, please comment and the issue will be reopened if appropriate.

closed 2019-06-17 by marcoesposito1988