IFL-CAMP/easy_handeye #34: How to save Calibration

runeg96, 2019-04-03:
Hey
Im using the eye in hand calibration and its working really well. But it seems like the save button doesn't work or its just me that cant find the yaml file with the calibration.
Hope you can help me :D

marcoesposito1988, 2019-04-08:
Hi @runeg96,

the file is saved into `~/.ros/easy_handeye/<value of the namespace argument of the calibration launch file>.yaml`.

However, this is an implementation detail. You should not need to extract the calibration result if you want to publish it in tf. You can use the publisher node, included in this package (as described in the [Publishing section of the README](https://github.com/IFL-CAMP/easy_handeye#publishing)). Everytime you restart the system, the publisher will automatically use the new value of the calibration.

marcoesposito1988, 2019-06-17:
Closing for inactivity. If the problem persists, please comment and the issue will be reopened if appropriate.

closed 2019-06-17 by marcoesposito1988