IFL-CAMP/easy_handeye #35: How to get rqt_easy_handeye Gui to run???

1gen100, 2019-04-10:
I have thi problem I'm not able to get my calibration GUI to pop up I only get this message, and tried to find a solution. Hope you can help:

[iiwa_kinect_handeyecalibration_eye_on_hand/namespace_dmri_w541_28293_3797793594356548394_rqt-18] process has died [pid 28475, exit code 1, cmd /opt/ros/kinetic/lib/rqt_gui/rqt_gui --clear-config --perspective-file /home/dmri/catkin_ws/src/easy_handeye/easy_handeye/launch/rqt_easy_handeye.perspective __name:=namespace_dmri_w541_28293_3797793594356548394_rqt __log:=/home/dmri/.ros/log/93e2877e-5b8d-11e9-acf4-54ee75886762/iiwa_kinect_handeyecalibration_eye_on_hand-namespace_dmri_w541_28293_3797793594356548394_rqt-18.log].
log file: /home/dmri/.ros/log/93e2877e-5b8d-11e9-acf4-54ee75886762/iiwa_kinect_handeyecalibration_eye_on_hand-namespace_dmri_w541_28293_3797793594356548394_rqt-18*.log

marcoesposito1988, 2019-04-16:
```
Could not import "pyqt" bindings of qt_gui_cpp library - so C++ plugins will not be available:
Traceback (most recent call last):
File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui_cpp/cpp_binding_helper.py", line 43, in 
from . import libqt_gui_cpp_sip
ImportError: dynamic module does not define module export function (PyInit_libqt_gui_cpp_sip)
```

Apparently on your computer there is a problem with the Qt Python dependencies. Have you used rosdep, as specified in the "Getting started" section of the README?

marcoesposito1988, 2020-03-14:
Closing for lack of feedback

closed 2020-03-14 by marcoesposito1988