IFL-CAMP/easy_handeye #37: Cannot detect the aruco marker and canTransform Error occurs

EvelynYibingWei, 2019-04-26:
Hi,
Hi, I'm using easy_handeye to calibrate Kinect v2 camera with the ur5 robot. 

I change the ur5_kinect_calibrate.lunch file as follow:

### <launch>
    <arg name="namespace_prefix" default="ur5_kinect_handeyecalibration" />
 
    <arg name="robot_ip" doc="The IP address of the UR5 robot" />
        
    <arg name="marker_size" doc="Size of the ArUco marker used, in meters" default="0.2" />
    <arg name="marker_id" doc="The ID of the ArUco marker used" default="100" />
    
    <include file="$(find kinect2_bridge)/launch/kinect2_bridge.launch" />
    
 
    
    <node name="aruco_tracker" pkg="aruco_ros" type="single">
        <remap from="/camera_info" to="/kinect2/hd/camera_info" />
        <remap from="/image" to="/kinect2/hd/image_color_rect" />
        <param name="image_is_rectified" value="true"/>
        <param name="marker_size"        value="$(arg marker_size)"/>
        <param name="marker_id"          value="$(arg marker_id)"/>
        <param name="reference_frame"    value="kinect2_rgb_optical_frame"/>
        <param name="camera_frame"       value="kinect2_rgb_optical_frame"/>
        <param name="marker_frame"       value="camera_marker" />
    </node>
 
    
    <include file="$(find ur_modern_driver)/launch/ur5_bringup.launch">
        <arg name="limited" value="true" />
        <arg name="robot_ip" value="192.168.137.2" />
    </include>
    <include file="$(find ur5_moveit_config)/launch/ur5_moveit_planning_execution.launch">
        <arg name="limited" value="true" />
    </include>
 
    
    <include file="$(find easy_handeye)/launch/calibrate.launch" >
        <arg name="namespace_prefix" value="$(arg namespace_prefix)" />
        <arg name="eye_on_hand" value="false" />
 
        <arg name="tracking_base_frame" value="kinect2_rgb_optical_frame" />
        <arg name="tracking_marker_frame" value="camera_marker" />
        <arg name="robot_base_frame" value="base_link" />
        <arg name="robot_effector_frame" value="wrist_3_link" />
 
        <arg name="freehand_robot_movement" value="false" />
        <arg name="robot_velocity_scaling" value="0.1" />
        <arg name="robot_acceleration_scaling" value="0.1" />
    </include>
</launch>

Just like issue#8 https://github.com/IFL-CAMP/easy_handeye/issues/8, I got the error "Error processing request: canTransform: source_frame camera_marker does not exist." when I try to take a sample.

 ['Traceback (most recent call last):\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 625, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/home/yibing/catkin_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_server.py", line 38, in take_sample\n    self.calibrator.take_sample()\n', '  File "/home/yibing/catkin_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_calibrator.py", line 140, in take_sample\n    transforms = self._get_transforms()\n', '  File "/home/yibing/catkin_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_calibrator.py", line 123, in _get_transforms\n    time = self._wait_for_transforms()\n', '  File "/home/yibing/catkin_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_calibrator.py", line 111, in _wait_for_transforms\n    self.listener.waitForTransform(self.tracking_base_frame, self.tracking_marker_frame, now, rospy.Duration(10))\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf/listener.py", line 76, in waitForTransform\n    raise tf2_ros.TransformException(error_msg or "no such transformation: \\"%s\\" -> \\"%s\\"" % (source_frame, target_frame))\n', 'TransformException: canTransform: source_frame camera_marker does not exist.\n']
Traceback (most recent call last):
  File "/home/yibing/catkin_ws/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_easy_handeye.py", line 104, in handle_take_sample
    sample_list = self.client.take_sample()
  File "/home/yibing/catkin_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_client.py", line 34, in take_sample
    return self.take_sample_proxy().samples
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 515, in call
    responses = transport.receive_once()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 727, in receive_once
    p.read_messages(b, msg_queue, sock)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 353, in read_messages
    self._read_ok_byte(b, sock)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 336, in _read_ok_byte
    raise ServiceException("service [%s] responded with an error: %s"%(self.resolved_name, str))
rospy.service.ServiceException: service [/ur5_kinect_handeyecalibration_eye_on_base/take_sample] responded with an error: error processing request: canTransform: source_frame camera_marker does not exist.

As showing in the  ur5_kinect_calibrate.lunch file, I added the camera_marker to tf tree. Why it  it doesn't exist? How to fix that?

Thanks very much!

EvelynYibingWei, 2019-04-26:
cannot detect the marker  even though the marker id is matched.
[image]

marcoesposito1988, 2019-04-29:
Hi @EvelynYibingWei,

I am sorry but I don't know why this happens. You might try asking in the [aruco_ros](https://github.com/pal-robotics/aruco_ros) repository.

EvelynYibingWei, 2019-04-29:
Thank you @marcoesposito1988

closed 2019-04-29 by EvelynYibingWei