IFL-CAMP/easy_handeye #38: arm[RRTConnectkConfigDefault]: Unable to sample any valid states for goal tree

kctoayo88, 2019-04-29:
Hi,

I am using ROS Kinetic, KInect V2 and Motoman MH5 Robot and trying to use the easy_handeye in my project.

When I click the check starting pose button, I got this error.

` [ERROR] [1556533936.017165846]: arm[RRTConnectkConfigDefault]: Unable to sample any valid states for goal tree`

I already check my urdf and didn't find any problem, and it works fine in demo.launch file.

[image]

Hope you cloud help me with this trouble.

Thanks a lot !

marcoesposito1988, 2019-04-29:
Hi @kctoayo88,

this means that the planner does not manage to move the robot in one direction starting from the current position of the robot.

You might try positioning the robot in a way such that its "hand" can rotate a bit about all axis, and translate in all directions, without reaching a joint limit.

The GUI automatically tries to move the robot in all directions before starting the procedure, to avoid giving an error in the middle of the calibration procedure (or at the end, which would be even more infuriating).

kctoayo88, 2019-04-30:
@marcoesposito1988 thanks for your reply.

I already tried to change the direction and angle of robot in many cases, but still can't work.
Is it any other solution could help me complete it?

I was trying to operate it manually, but result is poor.
Hope you could give me more tips of it.

Thanks a lot.

BeYoungs, 2019-05-22:
Hi
I'm using the easy_handeye_demo,when I click the check starting pose button, I got the error too.
[ERROR] [1558516337.830848905]: manipulator[RRTConnectkConfigDefault]: Unable to sample any valid states for goal tree
I have tried to put the hand which can rotate about all axis,but it doesn't work.
Hope you can give me some solutions.
Thanks a lot.

marcoesposito1988, 2020-03-14:
I think that the problem is not the orientation, but the fact that the robot is already at the edge of its workspace (i.e. it is almost fully elongated). 

You may try again moving the end effector towards the base of the robot.

closed 2020-03-17 by kctoayo88