IFL-CAMP/easy_handeye #39: Image View Problem

BobbleLaw, 2019-04-29:
Hi, 

Thank you for the work.

The latest code runs well on my computer without any error. But one weird problem bother me, I can't find anywhere to turn on Image View sub-window. It supposes to be under Plugins -> Visualisation, but all I have is a Plot. And I can rosrun image_view solely from a new command window to see aruco_result.

Do I need to setup somewhere in launch file, or install some package related to rqt?

Thanks.

marcoesposito1988, 2019-04-29:
Hi @BobbleLaw,

it sounds like you are missing some of the rqt packages. Maybe you did not install ROS through the `ros-DISTRO-desktop` package?
You should be able to fix the problem with `sudo apt install ros-melodic-rqt-image-view` (if you are using Melodic)

BobbleLaw, 2019-04-29:
Thank you so much. rqt-image-view is what I missed, that is a different package from image-view.

closed 2019-04-29 by BobbleLaw