IFL-CAMP/easy_handeye #40: A small question about the AR Marker

BeYoungs, 2019-05-23:
Hi
I'm brandly new about the calibration of eye to hand.
I just want to know,does the AR Marker is specially made ?
If it's maded specially,how can i made it?
Thanks

marcoesposito1988, 2019-05-23:
Hi @BeYoungs,

That is not necessary. You can print your marker with a standard inkjet printer.

What is important is that it is perfectly flat. You can glue it to a wooden or metal board, for example.

You must also double check the actual size of the marker after you print it (the printer might rescale the image automatically to fit it to the page).

BeYoungs, 2019-05-25:
First I'm very thankful for you advice.
I have checked the real marker size and keep it flat when I doing the calibration.
However,after doing the calibration for several times, including the camera calibration and some links being checked,the x and z direction calibration data is very good. But there is almost 5cm error in the real distance between the camera and ur3.It results in the object-picking cannot move to the right place.
Could you give me some solutions to improve the calibration results of the y direction(the calibration data between the camera and ur3).
I used the camera of ASUS Xtion pro,should I use other camera for a try?like kinect?
Thanks again.

marcoesposito1988, 2019-05-27:
Have you checked that the marker tracking is working correctly? This is a requirement for a good calibration.

You can check this without using the calibration, by moving the robot with the control panel, and comparing the Cartesian displacement of the end effector according to the robot panel and that of the marker according to the aruco marker. If you move the robot 5 cm up and the marker pose changes by 7 cm, you have a problem with the marker tracking (or a very big one with the robot, but that’s unlikely).

The hand-eye calibration uses external libraries for the computation, so if there is a problem it lies (usually) either in the configuration of the marker tracking, or in how you sample the data.

closed 2021-11-21 by marcoesposito1988