IFL-CAMP/easy_handeye #41: Error when running AUTO Movement

1gen100, 2019-06-24:
How can I get the Automatic movement to work when I get this fault?

[ERROR] [1561381645.473038930]: Robot model parameter not found! Did you remap 'robot_description'?
[FATAL] [1561381645.473096094]: Unable to construct robot model. Please make sure all needed information is on the parameter server.
PluginManager._load_plugin() could not load plugin "rqt_easy_handeye/Hand-eye Calibration automatic movement":
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler.py", line 99, in load
    self._load()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler_direct.py", line 54, in _load
    self._plugin = self._plugin_provider.load(self._instance_id.plugin_id, self._context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui_py/ros_py_plugin_provider.py", line 60, in load
    return super(RosPyPluginProvider, self).load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui/ros_plugin_provider.py", line 101, in load
    return class_ref(plugin_context)
  File "/home/dmri/catkin_ws/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_calibrationmovements.py", line 296, in __init__
    self._widget = CalibrationMovementsGUI()
  File "/home/dmri/catkin_ws/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_calibrationmovements.py", line 149, in __init__
    self.local_mover = CalibrationMovements(move_group_name, max_velocity_scaling, max_acceleration_scaling)
  File "/home/dmri/catkin_ws/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_calibrationmovements.py", line 26, in __init__
    self.mgc = MoveGroupCommander(move_group_name)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/moveit_commander/move_group.py", line 52, in __init__
    self._g = _moveit_move_group_interface.MoveGroupInterface(name, robot_description, ns)
RuntimeError: Unable to construct robot model. Please make sure all needed information is on the parameter server.

[calibration_mover-19] process has died [pid 8522, exit code 1, cmd /home/dmri/catkin_ws/src/easy_handeye/rqt_easy_handeye/scripts/rqt_calibrationmovements __name:=calibration_mover __log:=/home/dmri/.ros/log/fee5387c-9680-11e9-83d3-54ee75886762/calibration_mover-19.log].
log file: /home/dmri/.ros/log/fee5387c-9680-11e9-83d3-54ee75886762/calibration_mover-19*.log

marcoesposito1988, 2019-06-25:
```
[ERROR] [1561381645.473038930]: Robot model parameter not found! Did you remap 'robot_description'?
```
This means that you started the robot movement plugin before the robot model loaded. You should load the MoveIt! environment before easy_handeye.
For me it always worked by putting everything one after the other in a launch file, but if you somehow have a race condition you may want to first load MoveIt by hand, and then start easy_handeye.

1gen100, 2019-06-25:
Here is the launch file I used, and I'm starting the moveit before the easy_handeye.
 
`<launch>
    <arg name="namespace_prefix" default="easy_handeye" />
    <arg name="usr" default="$(optenv USER dmri)"/>
    <arg name="tool_file" default="/home/$(arg usr)/catkin_ws/src/easy_handeye/easy_handeye/config/tool_iiwa.json"/>

    <arg name="robot_ip" doc="The IP address of the iiwa robot" />

    <arg name="marker_size" default="0.1" doc="Size of the ArUco marker used, in meters" />
    <arg name="marker_id" default="582" doc="The ID of the ArUco marker used" />

    <arg name="marker_frame" default="marker" />
    <arg name="ref_frame" default="kinect2_rgb_optical_frame" />

  
  
    <include file="$(find kinect2_bridge)/launch/kinect2_bridge.launch"/>
    <arg name="img" default="hd" />

  
    <node name="aruco_tracker" pkg="aruco_ros" type="single">
        <remap from="/camera_info" to="/kinect2/$(arg img)/camera_info" />
        <remap from="/image" to="/kinect2/$(arg img)/image_color_rect" />
        <param name="image_is_rectified" value="true"/>
        <param name="marker_size"        value="$(arg marker_size)"/>
        <param name="marker_id"          value="$(arg marker_id)"/>
        <param name="reference_frame"    value="$(arg ref_frame)"/>
        <param name="camera_frame"       value="$(arg ref_frame)"/>
        <param name="marker_frame"       value="$(arg marker_frame)" />
    </node>

    <node pkg="aruco_ros" type="marker_publisher" name="aruco_marker_publisher">
        <remap from="/camera_info" to="/kinect2/$(arg img)/camera_info" />
        <remap from="/image" to="/kinect2/$(arg img)/image_color_rect" />
        <param name="image_is_rectified" value="True"/>
        <param name="marker_size"        value="$(arg marker_size)"/>
        <param name="reference_frame"    value="$(arg ref_frame)"/>   
        <param name="camera_frame"       value="$(arg ref_frame)"/>
    </node>

    
    <include file="$(find iiwa_moveit)/launch/moveit_planning_execution.launch">
      <arg name="model" value="iiwa14"/>
    	<arg name="sim" value="false"/>
    </include>

    
    <include file="$(find easy_handeye)/launch/calibrate.launch" >
        <arg name="namespace_prefix" value="$(arg namespace_prefix)" />
        <arg name="eye_on_hand" value="true" />

        <arg name="tracking_base_frame" value="$(arg ref_frame)" />
        <arg name="tracking_marker_frame" value="$(arg marker_frame)" />
        <arg name="robot_base_frame" value="iiwa_link_0" />
        <arg name="robot_effector_frame" value="iiwa_link_ee" />

        <arg name="freehand_robot_movement" value="false" />
        <arg name="robot_velocity_scaling" value="0.1" />
        <arg name="robot_acceleration_scaling" value="0.1" />
    </include>

</launch>
`

1gen100, 2019-06-25:
I have also tried runing two launch the first with the kinect, aruco and moveit and the second one with the easy_handeye, but still getting the same erro message.

marcoesposito1988, 2019-08-19:
Could you post the launch file that starts MoveIt! for the robot?

marcoesposito1988, 2020-03-14:
Closing for lack of feedback

closed 2020-03-14 by marcoesposito1988

huangjiancong1, 2021-11-13:
> starts MoveIt! for the robot?

Hi, @marcoesposito1988. 
In Kinova Kortex Gen3(https://github.com/kinovarobotics/ros_kortex), I believe the launch file of the driver already start the Moveit, please check this:
```
<?xml version="1.0"?>
<launch>
    
    <!-- Arm configuration -->
    <arg name="arm" default="gen3"/>
    <arg name="dof" default="7" if="$(eval arg('arm') == 'gen3')"/> <!-- Number of degrees of freedom of the arm -->
    <arg name="dof" default="6" if="$(eval arg('arm') == 'gen3_lite')"/> <!-- Number of degrees of freedom of the arm -->
    <arg name="vision" default="true"/> <!-- True if the arm has a Vision module -->
    <arg name="use_hard_limits" default="false"/> <!-- Set soft limits as hard limits (Gen3 only) -->

    <!-- Gripper configuration -->
    <!-- Default gripper for Gen3 is none, default gripper for Gen3 lite is gen3_lite_2f -->
    <arg name="gripper" default="robotiq_2f_85" if="$(eval arg('arm') == 'gen3')"/>
    <arg name="gripper" default="gen3_lite_2f" if="$(eval arg('arm') == 'gen3_lite')"/>

    <!-- Namespace -->
    <arg name="robot_name" default="my_$(arg arm)"/>
    <arg name="prefix" default=""/>

    <!-- Kortex API options -->
    <arg name="ip_address" default="192.168.2.10"/>
    <arg name="username" default="admin"/>
    <arg name="password" default="admin"/>
    <arg name="cyclic_data_publish_rate" default="40"/> <!--Hz-->
    <arg name="api_rpc_timeout_ms" default="2000"/> <!--milliseconds-->
    <arg name="api_session_inactivity_timeout_ms" default="35000"/> <!--milliseconds-->
    <arg name="api_connection_inactivity_timeout_ms" default="20000"/> <!--milliseconds-->

    <!-- Automatically start other modules -->
    <arg name="start_rviz" default="true"/>
    <arg name="start_moveit" default="true"/>

    <!-- Action server params -->
    <arg name="default_goal_time_tolerance" default="0.5"/> <!--seconds-->
    <arg name="default_goal_tolerance" default="0.005"/> <!--radians-->

    <group ns="$(arg robot_name)">
        <group if="$(eval not arg('prefix'))">
            <!-- Load the description for the robot -->
            <!-- Without gripper -->
            <param name="robot_description" command="$(find xacro)/xacro --inorder $(find kortex_description)/robots/$(arg arm).xacro dof:=$(arg dof) vision:=$(arg vision) sim:=false"
                if="$(eval not arg('gripper'))"/>
            <!-- With gripper -->
            <param name="robot_description" command="$(find xacro)/xacro --inorder $(find kortex_description)/robots/$(arg arm)_$(arg gripper).xacro dof:=$(arg dof) vision:=$(arg vision) sim:=false"
                unless="$(eval not arg('gripper'))"/>
        </group>
        <group unless="$(eval not arg('prefix'))">
            <!-- Load the description for the robot -->
            <!-- Without gripper -->
            <param name="robot_description" command="$(find xacro)/xacro --inorder $(find kortex_description)/robots/$(arg arm).xacro dof:=$(arg dof) vision:=$(arg vision) sim:=false prefix:=$(arg prefix)"
                if="$(eval not arg('gripper'))"/>
            <!-- With gripper -->
            <param name="robot_description" command="$(find xacro)/xacro --inorder $(find kortex_description)/robots/$(arg arm)_$(arg gripper).xacro dof:=$(arg dof) vision:=$(arg vision) sim:=false prefix:=$(arg prefix)"
                unless="$(eval not arg('gripper'))"/>
        </group>

        <!-- Start the kortex_driver node -->
        <node name="$(arg robot_name)_driver" pkg="kortex_driver" type="kortex_arm_driver" output="screen"> <!--launch-prefix="gdb -ex run args"-->
            <param name="sim" value="false"/>
            <param name="ip_address" value="$(arg ip_address)"/>
            <param name="username" value="$(arg username)"/>
            <param name="password" value="$(arg password)"/>
            <param name="cyclic_data_publish_rate" value="$(arg cyclic_data_publish_rate)"/>
            <param name="api_rpc_timeout_ms" value="$(arg api_rpc_timeout_ms)"/>
            <param name="api_session_inactivity_timeout_ms" value="$(arg api_session_inactivity_timeout_ms)"/>
[... 151 lines cut ...]
    </node>

    <!-- start easy_handeye -->
    <include file="$(find easy_handeye)/launch/calibrate.launch" >
        <arg name="namespace_prefix" value="$(arg namespace_prefix)" />
        <arg name="eye_on_hand" value="True" />
        <arg name="move_group" value="my_gen3"  doc="the name of move_group for the automatic robot motion with MoveIt!" />

        <arg name="tracking_base_frame" value="camera_color_frame" />
        <arg name="tracking_marker_frame" value="aruco_marker_frame" />
        <arg name="robot_base_frame" value="base_link" />
        <arg name="robot_effector_frame" value="tool_frame" />

        <arg name="freehand_robot_movement" value="false" />
        <arg name="robot_velocity_scaling" value="0.5" />
        <arg name="robot_acceleration_scaling" value="0.2" />
    </include>

</launch>
```

I still not figure out what is missed. Can you give me a hand?

Sincerely,
Jim

huangjiancong1, 2021-11-13:
> > starts MoveIt! for the robot?
> 
> Hi, @marcoesposito1988. In Kinova Kortex Gen3(https://github.com/kinovarobotics/ros_kortex), I believe the launch file(https://github.com/Kinovarobotics/ros_kortex/blob/noetic-devel/kortex_driver/launch/kortex_driver.launch) of the driver already start the Moveit, please check my local file:
> 
> ```
> <?xml version="1.0"?>
> <launch>
>     
>     
>     <arg name="arm" default="gen3"/>
>     <arg name="dof" default="7" if="$(eval arg('arm') == 'gen3')"/> 
>     <arg name="dof" default="6" if="$(eval arg('arm') == 'gen3_lite')"/> 
>     <arg name="vision" default="true"/> 
>     <arg name="use_hard_limits" default="false"/> 
> 
>     
>     
>     <arg name="gripper" default="robotiq_2f_85" if="$(eval arg('arm') == 'gen3')"/>
>     <arg name="gripper" default="gen3_lite_2f" if="$(eval arg('arm') == 'gen3_lite')"/>
> 
>     
>     <arg name="robot_name" default="my_$(arg arm)"/>
>     <arg name="prefix" default=""/>
> 
>     
>     <arg name="ip_address" default="192.168.2.10"/>
>     <arg name="username" default="admin"/>
>     <arg name="password" default="admin"/>
>     <arg name="cyclic_data_publish_rate" default="40"/> 
>     <arg name="api_rpc_timeout_ms" default="2000"/> 
>     <arg name="api_session_inactivity_timeout_ms" default="35000"/> 
>     <arg name="api_connection_inactivity_timeout_ms" default="20000"/> 
> 
>     
>     <arg name="start_rviz" default="true"/>
>     <arg name="start_moveit" default="true"/>
> 
>     
>     <arg name="default_goal_time_tolerance" default="0.5"/> 
>     <arg name="default_goal_tolerance" default="0.005"/> 
> 
>     <group ns="$(arg robot_name)">
>         <group if="$(eval not arg('prefix'))">
>             
>             
>             <param name="robot_description" command="$(find xacro)/xacro --inorder $(find kortex_description)/robots/$(arg arm).xacro dof:=$(arg dof) vision:=$(arg vision) sim:=false"
>                 if="$(eval not arg('gripper'))"/>
>             
>             <param name="robot_description" command="$(find xacro)/xacro --inorder $(find kortex_description)/robots/$(arg arm)_$(arg gripper).xacro dof:=$(arg dof) vision:=$(arg vision) sim:=false"
>                 unless="$(eval not arg('gripper'))"/>
>         </group>
>         <group unless="$(eval not arg('prefix'))">
>             
>             
>             <param name="robot_description" command="$(find xacro)/xacro --inorder $(find kortex_description)/robots/$(arg arm).xacro dof:=$(arg dof) vision:=$(arg vision) sim:=false prefix:=$(arg prefix)"
>                 if="$(eval not arg('gripper'))"/>
>             
>             <param name="robot_description" command="$(find xacro)/xacro --inorder $(find kortex_description)/robots/$(arg arm)_$(arg gripper).xacro dof:=$(arg dof) vision:=$(arg vision) sim:=false prefix:=$(arg prefix)"
>                 unless="$(eval not arg('gripper'))"/>
>         </group>
> 
> 
> 
>         
>         <node name="$(arg robot_name)_driver" pkg="kortex_driver" type="kortex_arm_driver" output="screen"> 
>             <param name="sim" value="false"/>
>             <param name="ip_address" value="$(arg ip_address)"/>
>             <param name="username" value="$(arg username)"/>
>             <param name="password" value="$(arg password)"/>
>             <param name="cyclic_data_publish_rate" value="$(arg cyclic_data_publish_rate)"/>
[... 186 lines cut ...]
'my_gen3'
ipdb> self.mgc = MoveGroupCommander(move_group_name
)
[ERROR] [1636773326.133558713]: Robot model parameter not found! Did you remap 'robot_description'?
[FATAL] [1636773326.133603257]: Unable to construct robot model. Please make sure all needed information is on the parameter server.
*** RuntimeError: Unable to construct robot model. Please make sure all needed information is on the parameter server.
ipdb> self.mgc = MoveGroupCommander(/my_gen3/move_g
roup)
*** SyntaxError: invalid syntax (<stdin>, line 1)
ipdb> self.mgc = MoveGroupCommander("/my_gen3/move_
group/")
[ERROR] [1636773353.374425029]: Robot model parameter not found! Did you remap 'robot_description'?
[FATAL] [1636773353.374485875]: Unable to construct robot model. Please make sure all needed information is on the parameter server.
*** RuntimeError: Unable to construct robot model. Please make sure all needed information is on the parameter server.
ipdb> self.mgc = MoveGroupCommander("/my_gen3/move_
group")
[ERROR] [1636773355.291994354]: Robot model parameter not found! Did you remap 'robot_description'?
[FATAL] [1636773355.292044698]: Unable to construct robot model. Please make sure all needed information is on the parameter server.
*** RuntimeError: Unable to construct robot model. Please make sure all needed information is on the parameter server.
ipdb> self.mgc = MoveGroupCommander("/move_group")
[ERROR] [1636773379.457427056]: Robot model parameter not found! Did you remap 'robot_description'?
[FATAL] [1636773379.457460482]: Unable to construct robot model. Please make sure all needed information is on the parameter server.
*** RuntimeError: Unable to construct robot model. Please make sure all needed information is on the parameter server.

```

marcoesposito1988, 2021-11-13:
Hi @huangjiancong1,

this is likely because the `move_group` argument that you pass to the `easy_handeye` launch file should be `arm`, not `my_gen3` (see here: https://github.com/Kinovarobotics/ros_kortex/blob/noetic-devel/kortex_move_it_config/gen3_move_it_config/config/7dof/gen3.srdf.xacro#L13)