IFL-CAMP/easy_handeye #45: chessboard target

Sinchiguano, 2019-08-09:
Hi,

Is it possible to execute the eye in hand calibration with a chessboard target? 

Best regards
Cesar

marcoesposito1988, 2019-08-19:
Hi @Sinchiguano,

you can use any tracking system that publishes the pose of the marker in tf. 

Your question has actually a perfect timing, since I just started working on Yet Another Aruco Wrapper because I couldn't find any that satisfied all of my needs (including detecting a charuco board). As soon as this is ready, I will publish it together with documentation on how to use it with `easy_handeye`.

marcoesposito1988, 2019-08-27:
I published a the new wrapper for ArUco, compatible also with ChArUco boards, as [easy_aruco](https://github.com/marcoesposito1988/easy_aruco). I will soon publish launch files to use it directly with `easy_handeye`, but you can already take a look at it if you wish.

kingjin94, 2019-10-28:
I think you can also use this package: http://wiki.ros.org/tuw_checkerboard

marcoesposito1988, 2019-11-01:
I evaluated it, but it only supports checkerboards and not markers. So I just wrote a wrapper that supports both.

Thanks anyway!

closed 2022-06-17 by marcoesposito1988