IFL-CAMP/easy_handeye #46: Eye-on-base calibration outcome - issue with publishing result

howard-wu-cam, 2019-08-28:
My eye-on-base calibration went really well, though a few little bumps, a **big THANKS** to the authors.

Here is my final yaml file:

    _eye_on_hand: false
    robot_base_frame: fixed_link
    tracking_base_frame: camera_base_link
    transformation: {qw: 0.7197444041856074, qx: -0.055009208985473554, qy: -0.07374779929687375,
  qz: -0.6881157182256235, x: -0.006299578434856121, y: 0.4156843936451614, z: 0.32690430193807757}_ 

The transform seems correct.

However, I got an issue when I published this transform.

In **/tf** , the transforms, as shown below, are wrong:
_transforms: 
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1567022266
        nsecs: 765403340
      frame_id: "fixed_link"
    child_frame_id: "camera_base_link"
    transform: 
      translation: 
        x: 1.0
        y: 1.0
        z: 1.0
      rotation: 
        x: 0.0
        y: 0.681638760023
        z: 0.0
        w: 0.731688868874_

However, in **/tf_static** , the transforms are correct, as shown below:

_transforms: 
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1567011306
        nsecs: 311048030
      frame_id: "fixed_link"
    child_frame_id: "camera_base_link"
    transform: 
      translation: 
        x: -0.00629957843486
        y: 0.415684393645
        z: 0.326904301938
      rotation: 
        x: -0.0550092089855
        y: -0.0737477992969
        z: -0.688115718226
        w: 0.719744404186_

Could you please advise where I get it wrong? thanks!

marcoesposito1988, 2019-08-29:
Hi @howard-wu-cam,

that looks like the dummy transform that is published during calibration by `calibrate.launch`. Do you still have the nodes from calibration running?

To check the calibration all you should have to do is to run `check_calibration.launch` with the appropriate parameters. To use the calibration in your setup, you should only include `publish.launch` with the same parameters (frames and calibration namespace).

howard-wu-cam, 2019-08-29:
Got another issue while doing the calibration. The sampling GUI seems not taking the correct transform.

Type: eye on base

My frame systems:
Tracking base frame (camera): camera_base_link
Tracking marker frame (marker): camera_marker
Robot base frame (world): fixed_link
Robot effector frame (hand): wrist_roll_link

The sample data I got in the sampling GUI: 
 1) 
 hand->world translation: 
  x: 0.320452250208
  y: -1.17708684084e-06
  z: -0.0842130576638
rotation: 
  x: 0.491126630004
  y: -0.491128434016
  z: -0.508718682479
  w: 0.508716813854 
 camera->marker translation: 
  x: 0.35861420596
  y: -0.00126671106251
  z: -0.0566779006961
rotation: 
  x: 0.523119863814
  y: -0.389996287897
  z: 0.451918495227
  w: 0.608291194397

Visual check my settings, the camera->marker translation seems correct, but the hand->world translation is not. 

Use tf_echo to check the hand -> world frame translation

hand->world transform is as follows:
 Translation: [0.000, 0.073, 0.323]
 Rotation: in Quaternion [-0.491, 0.491, 0.509, 0.509]
            in RPY (radian) [0.000, 1.536, 1.571]
            in RPY (degree) [0.006, 87.984, 90.006]
These values seem correct.

Can anyone advise where I got it wrong? Why did the sample GUI take different transform values? The Quaternion values seems similar between the sampled data and the tf_echo result, which make me feel more puzzled. Any suggestions? Thanks!

howard-wu-cam, 2019-08-29:
> Hi @howard-wu-cam,
>
> that looks like the dummy transform that is published during calibration by `calibrate.launch`. Do you still have the nodes from calibration running?
>
> To check the calibration [...]

Thank you, Marco @marcoesposito1988 ! Let me try this out and will report back here.

howard-wu-cam, 2019-08-29:
> Hi @howard-wu-cam,
>
> that looks like the dummy transform that is published during calibration by `calibrate.launch`. Do you still have the nodes from calibration running?
>
> To check the calibration [...]

Hi Marco @marcoesposito1988 , I tried the `publish.launch`, but can't establish the transform between `camera` and the `world (robot base)`. Checked both `/tf` and `/tf_static`, it seems transforms are established in `tf_static`, the outcome is as follows (using `rostopic echo /tf_static`)

transforms 

    header: 
      seq: 0
      stamp: 
        secs: 1567109987
        nsecs: 473409891
      frame_id: "fixed_link"
    child_frame_id: "camera_base_link"
    transform: 
      translation: 
        x: -0.00629957843486
        y: 0.415684393645
        z: 0.326904301938
      rotation: 
        x: -0.0550092089855
        y: -0.0737477992969
        z: -0.688115718226
        w: 0.719744404186
---

However, transform between `camera` and the `world (robot base)` doesn't exist in `/tf`, hence not able to form the frame tree. The result is shown below when I used `rosrun tf tf_echo fixed_link camera_base_link` to check the transform.

---
`Exception thrown:Could not find a connection between 'fixed_link' and 'camera_base_link' because they are not part of the same tree.Tf has two or more unconnected trees.`

---
Additionally, where can I find  `check_calibration.launch`? thanks!

Akumar201, 2022-09-29:
Hello, I am facing the same issue - transform between camera and the world (robot base) doesn't exist in /tf - 
, were you able to fix it ? 
[image]

marcoesposito1988, 2022-10-16:
This is likely because the camera is composed of multiple reference frames, organized as a tree. We can't publish the transform of a leaf reference frame (the optical center), but we should do that with the root instead (`camera_link`). So you should use that as camera frame.

Akumar201, 2022-10-17:
I resolved this issue by cloning the repository (https://github.com/IntelRealSense/realsense-ros)and adding the launch file (https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/rs_camera.launch) to calibrate.launch , this issue was while using Ubuntu 20.04 LTS