IFL-CAMP/easy_handeye #47: where is  the "camera_marker" defined?

superbigzz, 2019-09-08:
I dont know where the "camera_marker" is defined. Do I need to define it by myself?

YanhuiDua, 2019-09-09:
发自我的华为手机-------- 原始邮件 --------发件人： 冻鱼 <notifications@github.com>日期： 2019年9月8日周日 晚上9:11收件人： IFL-CAMP/easy_handeye <easy_handeye@noreply.github.com>抄送： Subscribed <subscribed@noreply.github.com>主    题： [IFL-CAMP/easy_handeye] where is  the "camera_marker" defined? (#47)I dont know where the "camera_marker" is defined. Do I need to define it by myself?

marcoesposito1988, 2019-09-09:
Hi @superbigbizz,

That is just a place holder for the actual tf frame published by the ROS driver of your tracking system. Since it is always different, all the launch files of easy_handeye allow you to specify the tf frames of interest via arguments. You can take a look at the readme to see how to include the launch files in your own to calibrate and then use the calibration.

closed 2020-03-14 by marcoesposito1988

mbajlo, 2020-03-18:
Hello I am confused with this part:

```
<!-- fill in the following parameters according to your tracking system's published tf frames -->
    <arg name="tracking_base_frame" value="/optical_origin"/>
    <arg name="tracking_marker_frame" value="/optical_target"/>

Where are `tracking base frame` and `tracking marker frame` are coming from?
Since the chessboard or the aruco are on the table, we should somehow estimate their transforms/poses?

```

marcoesposito1988, 2020-07-27:
Hello @mbajlo,

those are the names of the tf frames that your tracking software respectively uses for your camera and marker. These are either configured by you in the launch parameters of the ROS package that you are using to do the tracking, or documented by its developers if they are not configurable.

You don't need to estimate the initial pose. You just have to make sure that the tracking system and its software are properly calibrated and configured (intrinsic camera calibration, stereo camera calibration, marker size and ID, etc.)