IFL-CAMP/easy_handeye #48: A small question about handeye calibration for 4DoF robot

T2honda, 2019-09-10:
Hello

     Thank you very much for the great package.
     I want to make the eye to hand calibration for **4DoF ( x, y, z, yaw )** robot by the package, rather than the normal 6 DoF robot.
     I got the following figure by rviz: 
      
[image]

     and we can get the point cloud like that:
     
[image]

    I just wondering can I get calibration result for the 4DoF robot by the package ?
    For the case of 4DoF robot, could you please give some special suggestions for getting precision eye to hand calibration result ?

    I am very appreciated for your kindness help.

    Best Regards

marcoesposito1988, 2019-09-15:
Hello @T2honda,

from the images, I guess that the joint axes are parallel. In this case, I am afraid that the robot cannot rotate the marker in all 3 directions, so the calibration problem is undetermined and the algorithm cannot work. 

Furthermore, is the camera external to a mobile robot? In order for this to work, you would need to stop the robot base, and calibrate the arm just before picking up the object. Then perform the grasping without moving the robot base.

In your case I would try to measure the pose of the marker with respect to the end effector by hand.

T2honda, 2019-09-16:
Dear Marco

         Thank you so much for your kindness and useful suggestion.
         I understand that.

        Yes, the camera external to the robot arm. Actually, it is not a mobile robot, just a robot arm (URL about the robot arm: https://www.dobot.cc/dobot-magician/product-overview.html )
        
        However, I don't understand the last sentence "  **try to measure the pose of the marker with respect to the end effector by hand** " completely.  Since I just sticked the marker on the center of end effector, so they only exist difference in Z axis. 

        Do I understand it wrong?
        Could you please explain the last sentence again?
        I am very appreciated for your kindness help.

        Best Regards

marcoesposito1988, 2020-03-14:
Hi @T2honda,

I meant to directly measure the transformation by hand, such as with a ruler, or a caliper. If you have the URDF of the robot, and you know the location of the end effector frame on the flange, you may be able to measure this transformation with a precision that may suffice for your use case.

marcoesposito1988, 2020-10-05:
Closing for inactivity

closed 2020-10-05 by marcoesposito1988