IFL-CAMP/easy_handeye #50: no display of the GUI for rqt_easy_handeye

doublechenchen, 2019-11-10:
[image]

FelipeH92, 2019-11-19:
Any lucky with this?

marcoesposito1988, 2019-11-19:
This looks like a problem with your system: 
```ImportError: /usr/lib/x86_64-linux-gnu/libQt5Core.so.5: version 'Qt_5.13' not found```
Did you install ROS the standard way via APT? Did you at some point install Qt by hand?

FelipeH92, 2019-11-19:
I'm having a similar error. I get this

```
usage: rqt_gui [-b BINDING] [--clear-config] [-f] [--force-discover] [-h] [-l]
               [-ht] [-p PERSPECTIVE] [--perspective-file PERSPECTIVE_FILE]
               [--reload-import] [-s PLUGIN] [-t] [-v]
               [--args [PLUGIN_ARGS [PLUGIN_ARGS ...]]] [--list-perspectives]
               [--list-plugins] [--command-pid PID]
               [--command-start-plugin PLUGIN]
               [--command-switch-perspective PERSPECTIVE]
rqt_gui: error: unrecognized arguments: Workspaces/ROS Test Workspace/src/easy_handeye/easy_handeye/launch/rqt_easy_handeye.perspective
[ INFO] [1574194584.747175308]: AR tracker reconfigured: ENABLED 8.00 4.40 0.08 0.00
[ur5_kinect_handeyecalibration_eye_on_base/namespace_nascimento_Inspiron_7560_rqt-31] process has died [pid 10797, exit code 2, cmd /opt/ros/kinetic/lib/rqt_gui/rqt_gui --clear-config --perspective-file ~/home/nascimento/Projects/ROS Workspaces/ROS Test Workspace/src/easy_handeye/easy_handeye/launch/rqt_easy_handeye.perspective __name:=namespace_nascimento_Inspiron_7560_rqt __log:=/home/nascimento/.ros/log/74ce920c-0b09-11ea-9325-d80f99a74bf5/ur5_kinect_handeyecalibration_eye_on_base-namespace_nascimento_Inspiron_7560_rqt-31.log].
log file: /home/nascimento/.ros/log/74ce920c-0b09-11ea-9325-d80f99a74bf5/ur5_kinect_handeyecalibration_eye_on_base-namespace_nascimento_Inspiron_7560_rqt-31*.log

```

When I run `rosrun rqt_gui rqt_gui --clear-config` and try to open the perspective manually the gui crashes. I'm using Kinetic and Qt 5.5.1. Any clues?

FelipeH92, 2019-11-21:
I just updated to Melodic just to be sure but the bug persists. Even when initializing the node separately the gui doesn't start. I isolated the bug to this call on `handeye_client.py`

 ```
rospy.wait_for_service(hec.GET_SAMPLE_LIST_TOPIC)
 self.get_sample_proxy = rospy.ServiceProxy(hec.GET_SAMPLE_LIST_TOPIC, hec.srv.TakeSample)
```
From what I verified from `'calibrate.py'`, the server from `handeye_server.py` is initialized normally. Any ideas about what can it be? If I have any progress I'll report here.

marcoesposito1988, 2020-01-08:
Hi @FelipeH92,

looking here: `rqt_gui: error: unrecognized arguments: Workspaces/ROS Test Workspace/src/easy_handeye/easy_handeye/launch/rqt_easy_handeye.perspective` it seems like this is due to spaces in your path. Somewhere your path is split into multiple strings, and only the first one is taken as path. The rest is recognized as further (unexpected) arguments.

What is exactly the command that you are running? A fix may be possible on your side (and maybe only on your side). If you are running something like: 
```rosrun rqt_gui rqt_gui path/to/a directory with spaces within the name/containing/the/file.perspective```
you may avoid the error by enclosing the path within quotes:
```rosrun rqt_gui rqt_gui "path/to/a directory with spaces within the name/containing/the/file.perspective"```

If instead you are using my roslaunch files, I'll have to bang my head on it (of course any assistant headbangers would be welcome!)

marcoesposito1988, 2020-03-14:
Closing for lack of feedback

closed 2020-03-14 by marcoesposito1988