IFL-CAMP/easy_handeye #51: Error processing request: "camera_marker" passed to lookupTransform argument source_frame does not exist.

honggesmile, 2019-12-04:
I use the pachage easy_handeye.when I clicked "take sample",error  occured:"[ERROR] [1575441599.802618]: Error processing request: "camera_marker" passed to lookupTransform argument source_frame does not exist. "
have anyone help me ? Thanks very much.
note:I use realsense D435 and elfin Robot.
my launch file is in here:

<launch>
    <arg name="namespace_prefix" default="elfin5_realsense_handeyecalibration" />
 
    <arg name="robot_ip" doc="The IP address of the UR5 robot" />
		
    <arg name="marker_size" doc="Size of the ArUco marker used, in meters" default="0.1" />
    <arg name="marker_id" doc="The ID of the ArUco marker used" default="100" />
    
    
    
 
    
    <node name="aruco_tracker" pkg="aruco_ros" type="single">
        <remap from="/camera_info" to="/camera/color/camera_info" />
        <remap from="/image" to="/camera/color/image_raw" />
        <param name="image_is_rectified" value="true"/>
        <param name="marker_size"        value="$(arg marker_size)"/>
        <param name="marker_id"          value="$(arg marker_id)"/>
        <param name="reference_frame"    value="camera_color_optical_frame"/>
        <param name="camera_frame"       value="camera_color_optical_frame"/>
         
	<param name="marker_frame"       value="camera_marker" />
    </node>
 
    
    

    
        
        
         
 
    
    <include file="$(find easy_handeye)/launch/calibrate.launch" >
        <arg name="namespace_prefix" value="$(arg namespace_prefix)" />
        <arg name="eye_on_hand" value="true" />
 
        <arg name="tracking_base_frame" value="camera_color_optical_frame" />
        <arg name="tracking_marker_frame" value="camera_marker" />
        
        
	<arg name="robot_base_frame" value="elfin_base_link" />
 	<arg name="robot_effector_frame" value="elfin_end_link" />
 
        <arg name="freehand_robot_movement" value="false" />
        <arg name="robot_velocity_scaling" value="0.5" />
        <arg name="robot_acceleration_scaling" value="0.2" />
    </include>
 
</launch>

ayeyemeng, 2019-12-19:
Open the rqt-image-view through plugin while calibrating? However, this did not work with my things but you can make a try

desstiony, 2020-01-07:
Hello i encountered the same error, did you fix it？

marcoesposito1988, 2020-01-08:
Hi @honggesmile,

is the marker visible to the camera, and detected? Working marker tracking is a prerequisite for running the calibration.

desstiony, 2020-01-09:
> Hi @honggesmile,
> 
> is the marker visible to the camera, and detected? Working marker tracking is a prerequisite for running the calibration.

The marker is visible to the camera and is detected because you can see that the coordinate system is displayed
[Error screenshot](https://imgchr.com/i/lW3ulQ)
[Error screenshot](https://imgchr.com/i/lW3efS)

marcoesposito1988, 2020-01-09:
If you look at the logs, you can see that there is some problem with tf. It is discarding some data because it is apparently too old. Maybe some component is not using the ROS time for the timestamps?

In your case, the calibration can only work when this works:
`rosrun tf tf_echo camera_color_optical_frame camera_marker`
since it is what `easy_handeye` is doing in the background.

desstiony, 2020-01-09:
> If you look at the logs, you can see that there is some problem with tf. It is discarding some data because it is apparently too old. Maybe some component is not using the ROS time for the [...]

Thank you for your answer, I have solved my problem, I separate the work in the launch file to run separately; 1. start the robot arm, 2. run realsense, 3. start single under ArUco, 4. start easy_handeye
The tf of camera_color_frame and maker_frame were successfully established, but when single and easyhand under ArUco were started in a launch file, the tf relationship between camera_color_frame and maker_frame could not be established. I don't understand what caused it.
[xarm_realsense_calibration.launch](https://imgchr.com/i/lWjWFO)
[single.launch](https://imgchr.com/i/lWj2TK)

marcoesposito1988, 2020-03-14:
This is bizarre. You may try [my new ArUco wrapper](https://github.com/marcoesposito1988/easy_aruco), and see if that helps.

Exchizz, 2020-07-20:
I had the same error - Error processing request: "camera_marker" passed to lookupTransform argument source_frame does not exist.
When looking at the /aruco_tracker/result image, I could see the markerid of the marker I'm using. It was not the same as specified in my launch file.

I could have been nice if the tracker-node was saying something like "Detected marker 545, expected 3".....

marcoesposito1988, 2020-10-05:
This seems to be a configuration problem. Thanks for the observation, @Exchizz! I will try to do this in `easy_aruco`.

closed 2020-10-05 by marcoesposito1988