IFL-CAMP/easy_handeye #52: Questions about number of samples

double344931987, 2019-12-30:
Thanks for your code firstly!
I use your code with realsense in eye-on-base section.
During my experiment, if I took around 15 images, the results looks good but not exactly correct.
So, I take around 25, 30 and 40 samples, the results go worse. 
The result is confused me, in my opinion, the more samples, more data, the results more accurate, but actually not. 
Do you have this experience and know the problem is.
I am appreciate for your explanation firstly.

marcoesposito1988, 2020-01-08:
Hi @double344931987,

I am not familiar with the implementation nor really with the algorithm itself, so I do not feel confident enough to make any statement.

If you have an overdetermined system of equations given by the constraints formed by the samples, this may disturb the solver somehow. I plan to make it easy to switch the calibration algorithm and integrate new ones (maybe there are more robust alternatives), however the ViSP library has an excellent reputation.

You are sure that the marker tracking is visible and correctly detected when you take each sample? If the marker is temporarily not detected, the last position may still linger in tf until the data is discarded as too old. If outliers creep into your data, the result will be definitely wrong.

double344931987, 2020-01-08:
> Hi @double344931987,
>
> I am not familiar with the implementation nor really with the algorithm itself, so I do not feel confident enough to make any statement.
>
> If you have an overdetermined system [...]

double344931987, 2020-01-08:
Thank you for your reply firstly and I am looking forward to your new calibration algorithm.
I am sure I get the correct position and pictures. 
And I did many experiments (at least 3 times of each number of samples), all the results are the same as I mentioned before.
I read the algorithm of ViSP library but I cannot figure out why the results like this. 
If someone could prove the results or have same experience with me, I am happy to discuss it.

marcoesposito1988, 2020-10-05:
It may be worth checking that your hardware is working correctly. Maybe either the robot or the tracking system have a systematic error over the workspace.

One way to do it is to check that the relative movement of robot and tracking system coincide. Without performing the calibration you can't directly compare the motion, but you can check e.g. that the distance is the same. If you move the robot 10 mm, the tracking system should report exactly the same distance as well. If they don't agree, you may have a hardware problem.

closed 2021-11-21 by marcoesposito1988