IFL-CAMP/easy_handeye #55: Polaris_Spectra ndi is used as the camera to complete the calibration

TFLQW, 2020-04-26:
Hi~, First of all, Thanks for your contributuion of this package.
  Recently, I have got a equiment of ndi tracker which version is Polaris-Spectra, I have saw the picture in the instruction is ndi tracker, could give me some examples to show how to publish these two topics
<"/optical_origin"/> and <"/optical_target"/> using Polaris-Spectra. I haven't got any example showing the way to connect NDI through ROS.

marcoesposito1988, 2020-04-30:
Hi @TFLQW,

`optical_origin` and `optical_target` are not topics, but tf frames. What you need is to periodically obtain the value of the geometric transformation (rotation and translation, represented e.g. through a `geometry_msgs/Pose` message) between this two frames. That is, the transformation between the optical center of the camera and the reference frame attached to the marker; which is also the output of your tracking system.

So you need a ROS driver for your NDI tracking system, which publishes its input into `tf` as a transform between two tf reference frames. Unfortunately I could not find any with a simple search on GitHub, but you may be able to get started with this one: https://github.com/kuka-isir/polaris_sensor

However, that code does not publish the transform to tf, but publishes it on a topic. You will have to change the code to also publish the transform to tf (if you could fork the package on GitHub and make the changes public, that would be of course great and helpful for everyone else). You can take a look at my easy_aruco package to get started with this: in that package, I also publish the transform of the Aruco marker both to a topic and to tf.

TFLQW, 2020-05-02:
@marcoesposito1988 
Thans for your relpy, if I will try your suggestion, and If I have complete this project, I will publish my code.

TFLQW, 2020-09-04:
@marcoesposito1988 Hi~ Good news I have found the package [https://github.com/kuka-isir/polaris_sensor.git](url) to drive the ndi poloris device, and I found that one is too simple, so I have added some basic functions to match your easy_handeye package this is my modified version [https://github.com/TFLQW/polaris_sensor.git](url) in the branch of **tf_branch**

closed 2020-09-04 by TFLQW

marcoesposito1988, 2020-09-04:
That's great, thanks for writing!

At some point I will hopefully have the time to add a list of ROS tracking system drivers, and maybe add example configurations. In this case I will add a reference to your package.

nkusongzhichao, 2020-12-25:
Hi,
Thanks for your contributuion of this package, and I use the NDI Polaris Vega camera to calibration. It used  Ethernet rather than usb. How can I solve it? Thank you!