IFL-CAMP/easy_handeye #58: Regular WEbCamera

HamidROBO, 2020-08-26:
Hello,
First of all, I really thank you for publishing such as masterpiece. Second, I have a question about the camera type.
Does this published library work only for RGBD cameras? In other words, Can I use these codes for obtaining the transformation matrix between a web camera and a Robot base?
Regards,
Hamid

marcoesposito1988, 2020-08-26:
Hi @HamidROBO,

thank you for your kind words, I hope it continues to prove useful in the future!

The library is designed to work with all tracking systems that can publish the pose of the marker as a tf transform. AR marker tracking is an easy way to transform an RGBD or RGB camera into such a system, but you can also use optical stereo or electromagnetic tracking systems.

You can definitely use an RGB camera. You just have to make sure to perform first the [intrinsic calibration](http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration) (after you save it, it will be automatically used by easy_handeye).

Another thing you may have to pay attention to is that consumer webcams may do a lot of magic for you, which may make the calibration invalid. You may have to disable the autofocus function, and so on. Dedicated ROS camera drivers may expose an API to do it, or you can use standard v4l methods.

HamidROBO, 2020-08-26:
Dear Marco,
Thank you a lot for your kind and effective words. I will put them into
practice. They really were instructive for me.
Best wishes.
Hamid

marcoesposito1988, 2020-10-05:
Happy to help.. I hope everything goes smoothly!

closed 2020-10-05 by marcoesposito1988