IFL-CAMP/easy_handeye #59: [ERROR] [1600913615.065034]: Error processing request: "base_link" passed to lookupTransform argument target_frame does not exist.

chenguang3312, 2020-09-24:
`[INFO] [1600913605.063302]: Taking a sample...
[ERROR] [1600913615.065034]: Error processing request: "base_link" passed to lookupTransform argument target_frame does not exist. 
['Traceback (most recent call last):\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 625, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/home/robot/camera_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_server.py", line 38, in take_sample\n    self.calibrator.take_sample()\n', '  File "/home/robot/camera_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_calibrator.py", line 131, in take_sample\n    transforms = self._get_transforms()\n', '  File "/home/robot/camera_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_calibrator.py", line 116, in _get_transforms\n    rospy.Duration(10))\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_ros/buffer.py", line 87, in lookup_transform\n    return self.lookup_transform_core(target_frame, source_frame, time)\n', 'LookupException: "base_link" passed to lookupTransform argument target_frame does not exist. \n']
Traceback (most recent call last):
  File "/home/robot/camera_ws/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_easy_handeye.py", line 108, in handle_take_sample
    sample_list = self.client.take_sample()
  File "/home/robot/camera_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_client.py", line 34, in take_sample
    return self.take_sample_proxy().samples
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 515, in call
    responses = transport.receive_once()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 727, in receive_once
    p.read_messages(b, msg_queue, sock) 
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 353, in read_messages
    self._read_ok_byte(b, sock)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 336, in _read_ok_byte
    raise ServiceException("service [%s] responded with an error: %s"%(self.resolved_name, str))
rospy.service.ServiceException: service [/rgm_zed_handeyecalibration_eye_on_hand/take_sample] responded with an error: error processing request: "base_link" passed to lookupTransform argument target_frame does not exist.
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: 无效的参数
[rgm_zed_handeyecalibration_eye_on_hand/namespace_robot_16735_8950079066833835238_rqt-9] process has died [pid 16824, exit code -6, cmd /opt/ros/kinetic/lib/rqt_gui/rqt_gui --clear-config --perspective-file /home/robot/camera_ws/src/easy_handeye/easy_handeye/launch/rqt_easy_handeye.perspective __name:=namespace_robot_16735_8950079066833835238_rqt __log:=/home/robot/.ros/log/0be41b90-fe0b-11ea-bd62-aced5cc6308b/rgm_zed_handeyecalibration_eye_on_hand-namespace_robot_16735_8950079066833835238_rqt-9.log].
log file: /home/robot/.ros/log/0be41b90-fe0b-11ea-bd62-aced5cc6308b/rgm_zed_handeyecalibration_eye_on_hand-namespace_robot_16735_8950079066833835238_rqt-9*.log
ro
`

chenguang3312, 2020-09-24:
`<launch>
    <arg name="namespace_prefix" default="rgm_zed_handeyecalibration" />

    

    <arg name="marker_size" doc="Size of the ArUco marker used, in meters" />
    <arg name="marker_id" doc="The ID of the ArUco marker used" />

    

    
    <node name="aruco_tracker" pkg="aruco_ros" type="single">
        <remap from="/camera_info" to="/zedm/zed_node/rgb_raw/camera_info" />
        <remap from="/image" to="/zedm/zed_node/rgb_raw/image_raw_color" />
        <param name="image_is_rectified" value="true"/>
        <param name="marker_size"        value="0.15"/>
        <param name="marker_id"          value="100"/>
        <param name="reference_frame"    value="camera_fram"/>
        <param name="camera_frame"       value="camera_link"/>
        <param name="marker_frame"       value="camera_marker" />
    </node>

    
   <include file="$(find rgm_arm_moveit_config)/launch/moveit_planning_execution.launch">
    
   </include>

    
    <include file="$(find easy_handeye)/launch/calibrate.launch" >
        <arg name="namespace_prefix" value="$(arg namespace_prefix)" />
        <arg name="eye_on_hand" value="true" />

        <arg name="tracking_base_frame" value="camera_link" />
        <arg name="tracking_marker_frame" value="camera_marker" />
        <arg name="robot_base_frame" value="base_link" />
        <arg name="robot_effector_frame" value="tool0" />

        <arg name="freehand_robot_movement" value="false" />
        <arg name="robot_velocity_scaling" value="0.5" />
        <arg name="robot_acceleration_scaling" value="0.2" />
        <arg name="move_group" value="arm" />

    </include>

</launch>`

chenguang3312, 2020-09-24:
I have solved my problem
“DeserializationError cannot deserialize: unknown error handler name 'rosmsg' ”
The problem is that you are mixing older and newer Debian packages. you don't have the latest version of genpy yet. 
`sudo apt-get install ros-kinetic-genpy`

marcoesposito1988, 2020-10-05:
It looks like you are passing the wrong parameters, or you have a problem with MoveIt for your robot. 

`easy_handeye` is trying to sample the robot pose from `tf`, but apparently the `base_link` frame that was specified as `tf` frame of the robot base doesn't exist.

So either `base_link` is wrong and you should change the parameters that you pass to `calibrate.launch`, or you should check that your robot driver / MoveIt config starts correctly.

Does the problem still persist?

closed 2021-11-21 by marcoesposito1988