IFL-CAMP/easy_handeye #61: Hello,I use your new code,I met this trouble:an empty 'ns'

lijialei666, 2020-10-08:
[image]

marcoesposito1988, 2020-10-08:
Thank you for reporting this!

Can you please try adding `<arg name="move_group_namespace" value="/">` to your inclusion of `calibrate.launch`?

If this works, I will set this as the default value.

lijialei666, 2020-10-08:
the problem is resolved,but I met new problems.
[image]
[image]
[image]

lijialei666, 2020-10-08:
I need to solve this problem urgently. It has been bothering me for seven days,thank you very much.

marcoesposito1988, 2020-10-08:
I would like to help you, but those problems look like they are not related to `easy_handeye` so I don't know if I can.

The first and third screenshot seem to point to a problem with your Qt installation. Does `rviz` start on your system?

The second says that `pyassimp` can't be imported by python, and it has a link to a known issue with MoveIt. I guess the solution should be there.

lijialei666, 2020-10-08:
Thank you very much! I will try it.

QRayMoor, 2020-10-25:
> Thank you very much! I will try it.

Hi there, I have the same problems you met. Have you solved it?
1. Where to add this `<arg name="move_group_namespace" value="/">` in `calibrate.launch`?
2. And how to solve the second screenshot problem ?

lijialei666, 2020-10-25:
> > Thank you very much! I will try it.
> 
> Hi there, I have the same problems you met. Have you solved it?
> 
> 1. Where to add this `<arg name="move_group_namespace" value="/">` in `calibrate.launch`?
> 2. And how to solve the second screenshot problem ?

Hello, the first problem I have solved in calibrate.launch. The second problem I have not solved.

marcoesposito1988, 2020-10-28:
Have you tried `sudo apt install python-pyassimp`?

lijialei666, 2020-10-29:
> Have you tried `sudo apt install python-pyassimp`?

No, I have not tried it. But I've completed the calibration successfully by using old version easy_hand_eye.

marcoesposito1988, 2020-10-29:
That's very interesting.. if that happens, then it should happen with both versions.

Which ROS distro are you using?

lijialei666, 2020-10-29:
> That's very interesting.. if that happens, then it should happen with both versions.
> 
> Which ROS distro are you using?

Kinetic

marcoesposito1988, 2020-10-30:
Ok, so it's definitely python 2.

I will try to reproduce the issue on my side

lijialei666, 2020-10-31:
> I'm very sorry about this, I forgot to change back the shebang in the scripts. I just fixed this [now](https://github.com/IFL-CAMP/easy_handeye/commit/82576de5ddd302ee89592c2b5ea4073e91cdf288).
> 
> If you pull from the master branch, the new version should now work on all ROS distros.

Ok

closed 2021-11-21 by marcoesposito1988