IFL-CAMP/easy_handeye #62: i don't run rqt_easy_handeye/Hand-eye Calibration automatic movement ,hope your help

aoligeihahah, 2020-10-11:
PluginManager._load_plugin() could not load plugin "rqt_easy_handeye/Hand-eye Calibration automatic movement":
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler.py", line 102, in load
    self._load()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler_direct.py", line 55, in _load
    self._plugin = self._plugin_provider.load(self._instance_id.plugin_id, self._context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 72, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 72, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui_py/ros_py_plugin_provider.py", line 60, in load
    return super(RosPyPluginProvider, self).load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 72, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui/ros_plugin_provider.py", line 101, in load
    return class_ref(plugin_context)
  File "/home/he/catkin_ws/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_calibrationmovements.py", line 306, in __init__
    self._widget = CalibrationMovementsGUI()

aoligeihahah, 2020-10-11:
[image]
[image]

aoligeihahah, 2020-10-11:
[image]
[image]

marcoesposito1988, 2020-10-11:
I can't explain this behavior; I opened an [issue](https://github.com/ros-planning/moveit/issues/2364) in the MoveIt repo to ask for clarification.

This issue will happen in most cases, since usually the `move_group` action is located in the global namespace. As a workaround, you can include the launch file for your robot in a namespace, then pass that namespace as argument to `easy_handeye`:

```
<!-- note the ns="..." -->
<include ns="/my_robot" file="$(find ur10_moveit_config)/launch/..."/>

<!-- add the same namespace as move_group_namespace argument -->
<include file="$(find easy_handeye)/launch/calibrate.launch" >
...
        <arg name="move_group_namespace" value="/my_robot" />
...
</include>
```

aoligeihahah, 2020-10-12:
Thank you for your help, but I'm showing this problem。
[image]

aoligeihahah, 2020-10-12:
I have added it as shown in the figure below.
[image]

marcoesposito1988, 2020-10-12:
You are right, that workaround is not enough.

I should have managed to fix the problem in a way that doesn't require a workaround. I updated the code on the master branch; I hope this solves your problem. To test for this, I created a [version of the demo](https://github.com/marcoesposito1988/easy_handeye_demo/tree/master/launch/universal_robot) that uses the Universal Robot UR5.

aoligeihahah, 2020-10-19:
Hello, I have updated the code, but I have some problems with the pictures. I hope you can help me
[image]
[image]
[image]

marcoesposito1988, 2020-10-19:
There is a problem with your system: it looks like PyQt5 is missing

aoligeihahah, 2020-11-01:
[image]
[image]
[image]
[image]
[image]
[image]

aoligeihahah, 2020-12-01:
I have run successfully, but there is another problem

closed 2020-12-01 by aoligeihahah