IFL-CAMP/easy_handeye #63: The robot coordinate is not match the UR5's controller screen？

zhangxiaodi, 2020-10-16:
hi author, when I use this promising project to calibrate the ur5 (eye-to-hand), I found some interesting things. we print every calibrated sample. 
The rob is not matched to UR5's controller screen. So maybe I don't understand the frame real means what. can you give me some advice about this?
The rob is followed here:
 'rob:', header:
  seq: 0
  stamp:
        xxxxxx
  frame_id: "tool0"
child_frame_id: "base_link"
transform:
   translation:
        x : 0.63651359237
        y: -0.317187866488
        z:0.0533706444122
    rotation:
        x: 0.108439787775
        y:0.941737078689
        z:-0.0962134564323
        w:-0.30350462575

The UR% screen relative coordinate control:
        x  -0.64810
        y  -0.18274
        z  0.25704
        RX  3.1090   rad
        RY  -0.4966   rad
        RZ   -1.0270   rad
So I have some doubt about rob coordinate. Where is it from?

zhangxiaodi, 2020-10-16:
[image]

marcoesposito1988, 2020-10-16:
Hi @zhangxiaodi,

yes, I remember some weird thing with the UR5's tf frames. There is a weird rotation between the base reference frame used by the robot's internal controller and that used by ROS/MoveIt. You may have to use `base` instead of `base_link`.

However, I haven't used this robot in a while and things may have changed (see for example https://github.com/ros-industrial/universal_robot/pull/505).

zhangxiaodi, 2020-10-22:
Thank you for your kind answer. Maybe i should change the coordinate base_link to base.

And this weird thing is really to solve.   
  
    <link name="${prefix}base"/>
    <joint name="${prefix}base_link-base_fixed_joint" type="fixed">
      
      <origin xyz="0 0 0" rpy="0 0 ${-pi}"/>
      <parent link="${prefix}base_link"/>
      <child link="${prefix}base"/>
    </joint>

closed 2020-10-22 by zhangxiaodi