IFL-CAMP/easy_handeye #64: when i click "next pose" and then "plan", the program is broken.

makangzhe, 2020-10-21:
i am  sure the marker in the center of picture .   the error like this ,can you help me ?
 

`started roslaunch server http://192.168.0.133:45373/

SUMMARY
========

PARAMETERS
 * /calibration_mover/max_acceleration_scaling: 0.2
 * /calibration_mover/max_velocity_scaling: 0.5
 * /calibration_mover/move_group: manipulator_i5
 * /calibration_mover/rotation_delta_degrees: 25
 * /calibration_mover/translation_delta_meters: 0.1
 * /easy_handeye_eye_on_hand/eye_on_hand: True
 * /easy_handeye_eye_on_hand/robot_base_frame: base_link
 * /easy_handeye_eye_on_hand/robot_effector_frame: wrist3_Link
 * /easy_handeye_eye_on_hand/tracking_base_frame: camera_link
 * /easy_handeye_eye_on_hand/tracking_marker_frame: camera_marker
 * /rosdistro: kinetic
 * /rosversion: 1.12.16

NODES
  /easy_handeye_eye_on_hand/
    hand_eye_connector (easy_handeye/calibrate.py)
    hand_eye_solver (visp_hand2eye_calibration/visp_hand2eye_calibration_calibrator)
    namespace_StudyRos_16678_2847760988164325460_rqt (rqt_gui/rqt_gui)
  /
    calibration_mover (rqt_easy_handeye/rqt_calibrationmovements)
    dummy_handeye (tf/static_transform_publisher)
    rviz_StudyRos_16678_3072753249223410377 (rviz/rviz)

ROS_MASTER_URI=http://192.168.0.154:11311

process[dummy_handeye-1]: started with pid [16690]
process[easy_handeye_eye_on_hand/hand_eye_solver-2]: started with pid [16691]
process[easy_handeye_eye_on_hand/hand_eye_connector-3]: started with pid [16692]
process[rviz_StudyRos_16678_3072753249223410377-4]: started with pid [16696]
process[easy_handeye_eye_on_hand/namespace_StudyRos_16678_2847760988164325460_rqt-5]: started with pid [16698]
process[calibration_mover-6]: started with pid [16699]
[ INFO] [1603293281.419253798]: rviz version 1.12.17
[ INFO] [1603293281.419420916]: compiled against Qt version 5.5.1
[ INFO] [1603293281.419435071]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1603293281.953197699]: Stereo is NOT SUPPORTED
[ INFO] [1603293281.953309713]: OpenGl version: 3 (GLSL 1.3).
arguments:  Namespace(quiet=False)
unknowns:  []
[ INFO] [1603293285.798004331]: Loading robot model 'aubo_i5'...
[ WARN] [1603293285.798078636]: Skipping virtual joint 'base_link' because its child frame 'base_link' does not match the URDF frame 'world'
[ INFO] [1603293285.798102005]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1603293286.259267440]: Loading robot model 'aubo_i5'...
[ WARN] [1603293286.260014952]: Skipping virtual joint 'base_link' because its child frame 'base_link' does not match the URDF frame 'world'
[ INFO] [1603293286.260172351]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1603293287.524919067]: Starting scene monitor
[ INFO] [1603293287.547127945]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1603293288.876713947]: Constructing new MoveGroup connection for group 'manipulator_i5' in namespace ''
[ WARN] [1603293289.180472308]: Interactive marker 'EE:goal_wrist3_Link' contains unnormalized quaternions. This warning will only be output once but may be true for others; enable DEBUG messages for ros.rviz.quaternions to see more details.
[ INFO] [1603293290.173379487]: Ready to take commands for planning group manipulator_i5.
[ INFO] [1603293290.173676787]: Looking around: no
[ INFO] [1603293290.173788322]: Replanning: no
Traceback (most recent call last):
  File "/home/mkz/catkin_ws/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_calibrationmovements.py", line 271, in handle_plan
    self.local_mover.fallback_joint_limits):  # TODO: sort out this limits story
  File "/home/mkz/catkin_ws/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_calibrationmovements.py", line 129, in is_crazy_plan
    abs_rot_per_joint = CalibrationMovements.rot_per_joint(plan)
  File "/home/mkz/catkin_ws/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_calibrationmovements.py", line 119, in rot_per_joint
    raise ValueError
ValueError
[calibration_mover-6] process has died [pid 16699, exit code -11, cmd /home/mkz/catkin_ws/src/easy_handeye/rqt_easy_handeye/scripts/rqt_calibrationmovements __name:=calibration_mover __log:=/home/mkz/.ros/log/99e0e17a-13ab-11eb-b077-00026fbe871c/calibration_mover-6.log].
log file: /home/mkz/.ros/log/99e0e17a-13ab-11eb-b077-00026fbe871c/calibration_mover-6*.log
[ERROR] [1603293581.143441966]: [getPublishedTopics] Failed to contact master at [192.168.0.154:11311].  Retrying..`

makangzhe, 2020-10-21:
i am using old version .

makangzhe, 2020-10-21:
this is my launch file

<?xml version="1.0"?>
<launch>
    <arg name="namespace_prefix" default="auboi5_realsense_handeyecalibration" />

    <arg name="robot_ip" doc="The IP address of the UR5 robot" default="192.168.0.100" />

    <arg name="marker_size" doc="Size of the ArUco marker used, in meters" default="0.08"/>
    
    <arg name="marker_id" doc="The ID of the ArUco marker used" default="100"/>

    
    <node name="aruco_tracker" pkg="aruco_ros" type="single">
        <remap from="/camera_info" to="/camera/color/camera_info" />
        <remap from="/image" to="/camera/color/image_raw" />
        <param name="image_is_rectified" value="true"/>
        <param name="marker_size"        value="$(arg marker_size)"/>
        <param name="marker_id"          value="$(arg marker_id)"/>
        <param name="reference_frame"    value="camera_link"/>
        <param name="camera_frame"       value="camera_color_optical_frame"/>
        <param name="marker_frame"       value="camera_marker" />
    </node>

    
    <include file="$(find aubo_i5_moveit_config)/launch/moveit_planning_execution.launch">
        <arg name="sim" value="false" />
        <arg name="robot_ip" value="192.168.0.100" />
    </include>

    
    <include file="$(find easy_handeye)/launch/calibrate.launch" >
        <arg name="namespace_prefix" value="$(arg namespace_prefix)" />
        <arg name="eye_on_hand" value="true" />

        <arg name="tracking_base_frame" value="camera_link" />
        <arg name="tracking_marker_frame" value="camera_marker" />
        <arg name="robot_base_frame" value="base_link" />
        <arg name="robot_effector_frame" value="wrist3_Link" />

        <arg name="freehand_robot_movement" value="false" />
        <arg name="robot_velocity_scaling" value="0.5" />
        <arg name="robot_acceleration_scaling" value="0.2" />
    </include>
</launch>

makangzhe, 2020-10-22:
@marcoesposito1988

marcoesposito1988, 2020-10-23:
Hello @makangzhe,

I added a configuration for the aubo i5 robot to easy_handeye_demo; you can find it [here](https://github.com/marcoesposito1988/easy_handeye_demo/tree/master/launch/aubo_robot).

I did not have any problems calibrating with the simulator. Can you also try it? (I used the latest master of `easy_handeye`)

If you have problems with the simulator as well, please try another starting position for the robot.

Otherwise, please check that you can move the robot with MoveIt and pay attention to the `easy_handeye` logs, to see if there are problems in the connection between `easy_handeye` and MoveIt.

lucamarchionna, 2021-07-06:
Hi everyone, I am experiencing the same problem. Basically, I have launched all the needed files to perform calibration, according to the step you described. However, when I click on the 'check starting pose' button, it returns "Cannot calibrate from current position". For sure, I am skipping some important steps but I don't know which one. In addition, I am able to successfully move the robot within RViZ.
Below, you can find the launch file and a photo of the result. 

Photo:
[image]

Launch file:
[image]

marcoesposito1988, 2021-07-06:
Sorry, just saw this; the answer is [here](https://github.com/marcoesposito1988/easy_handeye_demo/issues/2#issuecomment-874849275)

lucamarchionna, 2021-07-06:
Thank you for the reply @marcoesposito1988 . I have just checked different random positions in MoveIT and the check starting pose always fails. 

I am performing the setup in simulation moving the camera by hand- not together with robot end-effector- just to see if a starting pose is available. By the way, what is the argument to reduce the rotations and translation?

In addition, I saw that 'freehand_robot_movement' is present, but I don't know how it works. Can you explain what this is, please? Thank you in advance.

marcoesposito1988, 2021-07-07:
You can find the arguments related to the automatic robot movement [here](https://github.com/IFL-CAMP/easy_handeye/blob/master/easy_handeye/launch/calibrate.launch#L10)

`freehand_robot_movement` will disable the robot movements GUI, so it won't try to connect to an existing MoveIt group and it won't clutter your logs with errors if it doesn't succeed. You still need a `robot_state_publisher` running to have the pose of the base and effector frames in `tf` for the whole thing to work though.

`rotation_delta_degrees` and `translation_delta_meters` specify how far apart the calibration poses should be from the starting pose. 

Is the description of your robot public? If so, I can maybe take a look at it and try to figure out why the IK is failing

lucamarchionna, 2021-07-07:
Great documentation for this package, congrats. Thus, the /joint_states publishes its pose and seems to work. I cannot detect which could be the error although it is still present. It would be great if you can check. I am using e.Do robot, here the robot description package: https://github.com/Pro/edo_gripper_moveit

Thanks again for your support.