IFL-CAMP/easy_handeye #65: Can this package calibrate franka panda robot arm?

saprrow, 2020-10-28:


marcoesposito1988, 2020-10-28:
Hi @saprrow,

yes, it can calibrate any robot arm with MoveIt support.

I added an example to [easy_handeye_demo](https://github.com/marcoesposito1988/easy_handeye_demo/tree/master/launch/franka_panda), which you can take a look at to. You can just copy the launch files for the part regarding the robot. You will have to replace the tracking simulator with your actual tracking system.

saprrow, 2020-10-29:
Okay! Thank you so much!

saprrow, 2020-10-29:
> Hi @saprrow,
>
> yes, it can calibrate any robot arm with MoveIt support.
>
> I added an example to [...]

> Hi @saprrow,
>
> yes, it can calibrate any robot arm with MoveIt support.
>
> I added an example to [...]

Okay. Thank you so much!

marcoesposito1988, 2020-10-30:
You're welcome! Happy calibrating!

closed 2020-10-30 by marcoesposito1988