IFL-CAMP/easy_handeye #69: My rqt_ easy_handeye There is no menu in handeye interface. ask your help

aoligeihahah, 2020-12-01:
[image]
If I click takesample, I will report an error. I don't know the reason and ask for your help

 Error processing request: "camera_marker" passed to lookupTransform argument source_frame does not exist. 
['Traceback (most recent call last):\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 625, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/home/he/catkin_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_server.py", line 61, in take_sample\n    self.sampler.take_sample()\n', '  File "/home/he/catkin_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_sampler.py", line 88, in take_sample\n    transforms = self._get_transforms()\n', '  File "/home/he/catkin_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_sampler.py", line 78, in _get_transforms\n    Duration(10))\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_ros/buffer.py", line 87, in lookup_transform\n    return self.lookup_transform_core(target_frame, source_frame, time)\n', 'LookupException: "camera_marker" passed to lookupTransform argument source_frame does not exist. \n']
Traceback (most recent call last):
  File "/home/he/catkin_ws/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_easy_handeye.py", line 118, in handle_take_sample
    sample_list = self.client.take_sample()
  File "/home/he/catkin_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_client.py", line 69, in take_sample
    return self.take_sample_proxy().samples
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 515, in call
    responses = transport.receive_once()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 728, in receive_once
    p.read_messages(b, msg_queue, sock) 
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 353, in read_messages
    self._read_ok_byte(b, sock)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 336, in _read_ok_byte
    raise ServiceException("service [%s] responded with an error: %s"%(self.resolved_name, str))
rospy.service.ServiceException: service [/ur10_kinect2_handeyecalibration_eye_on_base/take_sample] responded with an error: error processing request: "camera_marker" passed to lookupTransform argument source_frame does not exist.
[ur10_kinect2_handeyecalibration_eye_on_base/namespace_he_OptiPlex_5040_17632_2104357553874637777_rqt-17] process has died [pid 17692, exit code 1, cmd /home/he/catkin_ws/src/easy_handeye/rqt_easy_handeye/scripts/rqt_easy_handeye __name:=namespace_he_OptiPlex_5040_17632_2104357553874637777_rqt __log:=/home/he/.ros/log/15dc51b4-33f6-11eb-8f33-484d7ea54583/ur10_kinect2_handeyecalibration_eye_on_base-namespace_he_OptiPlex_5040_17632_2104357553874637777_rqt-17.log].
log file: /home/he/.ros/log/15dc51b4-33f6-11eb-8f33-484d7ea54583/ur10_kinect2_handeyecalibration_eye_on_base-namespace_he_OptiPlex_5040_17632_2104357553874637777_rqt-17*.log

marcoesposito1988, 2020-12-01:
Hello @aoligeihahah,

the sampler crashed, because the tf frame `camera_marker` that is configured in your launch file as the marker frame doesn't exist. You will have to change that parameter for your particular setup.

Since you are calibrating a kinect, I guess that you are using an AR marker tracking library, like ArUco. That library should publish the pose of the marker in tf, as a transform from the camera to the marker. You will have to insert the tf frame of the marker as your `tracking_marker` frame when you include or launch `easy_handeye/calibrate.launch`

aoligeihahah, 2020-12-02:
Hello，I want to open the rqt menu bar and select plugins - visualization - image view, as shown in the figure below,
But I have no menu bar to choose from. What's the reason[image]

aoligeihahah, 2020-12-02:
My system is:
ubantu16.04+ros-kinetic
camera：Kinect2

lyh458, 2020-12-03:
> Hello，I want to open the rqt menu bar and select plugins - visualization - image view, as shown in the figure below,
> But I have no menu bar to choose from. What's the reason[image]

Simply, you can just press “Add” button on your first rviz GUI, then select "image" plugin and choose the topic you want to show.
[image]
Another way is run ``rqt``, after that, a separated window will be showed and  then choose the topic you want to be showed (I think the way you showed in the quoted picture is close to this, but I don't know how to combine them in one window.)

aoligeihahah, 2020-12-03:
Thank you. I did as you said. My image didn't receive the image.

I'm very strange, if I just run aruco_ ros's single. Launch will get a color image, but run ur10_ kinect2_ calibration.launch After that, I tried rosrun image_ view image_ view image:=/aruco_ Single / result, the image will not be obtained and the window will not respond, as shown in the following figure:
[image]

aoligeihahah, 2020-12-03:
he@he-OptiPlex-5040:~$ cd catkin_ws
he@he-OptiPlex-5040:~/catkin_ws$ source devel/setup.bash
he@he-OptiPlex-5040:~/catkin_ws$  roslaunch easy_handeye ur10_kinect2_calibration.launch
... logging to /home/he/.ros/log/f3c9a586-3571-11eb-9154-484d7ea54583/roslaunch-he-OptiPlex-5040-16090.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

/home/he/.local/lib/python2.7/site-packages/pkg_resources/py2_warn.py:21: UserWarning: Setuptools will stop working on Python 2
************************************************************
You are running Setuptools on Python 2, which is no longer
supported and
>>> SETUPTOOLS WILL STOP WORKING <<<
in a subsequent release (no sooner than 2020-04-20).
Please ensure you are installing
Setuptools using pip 9.x or later or pin to `setuptools<45`
in your environment.
If you have done those things and are still encountering
this message, please follow up at
https://bit.ly/setuptools-py2-warning.
************************************************************
  sys.version_info < (3,) and warnings.warn(pre + "*" * 60 + msg + "*" * 60)
/home/he/.local/lib/python2.7/site-packages/pkg_resources/py2_warn.py:21: UserWarning: Setuptools will stop working on Python 2
************************************************************
You are running Setuptools on Python 2, which is no longer
supported and
>>> SETUPTOOLS WILL STOP WORKING <<<
in a subsequent release (no sooner than 2020-04-20).
Please ensure you are installing
Setuptools using pip 9.x or later or pin to `setuptools<45`
in your environment.
If you have done those things and are still encountering
this message, please follow up at
https://bit.ly/setuptools-py2-warning.
************************************************************
  sys.version_info < (3,) and warnings.warn(pre + "*" * 60 + msg + "*" * 60)
WARN: unrecognized 'node' tag in <include> tag
/opt/ros/kinetic/lib/python2.7/dist-packages/xacro/__init__.py:102: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  return yaml.load(f)
/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/loader.py:409: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  data = yaml.load(text)
started roslaunch server http://he-OptiPlex-5040:39483/

SUMMARY
========

PARAMETERS
 * /controller_stopper/consistent_controllers: ['joint_state_con...
 * /easy_handeye_calibration_server_robot/calibration_namespace: ur10_kinect2_hand...
 * /force_torque_sensor_controller/publish_rate: 125
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /hardware_control_loop/loop_hz: 125
 * /hardware_interface/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 125
 * /joint_state_controller/type: joint_state_contr...
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/he/catkin_w...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 12.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: False
[... 554 lines cut ...]
[FATAL] [1607013303.924309265]: Failed to load nodelet '/kinect2_points_xyzrgb_qhd` of type `depth_image_proc/point_cloud_xyzrgb` to manager `kinect2'
[FATAL] [1607013303.924481145]: Failed to load nodelet '/kinect2_points_xyzrgb_sd` of type `depth_image_proc/point_cloud_xyzrgb` to manager `kinect2'
[FATAL] [1607013303.924484701]: Failed to load nodelet '/kinect2_bridge` of type `kinect2_bridge/kinect2_bridge_nodelet` to manager `kinect2'
[kinect2-1] process has died [pid 16110, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=kinect2 __log:=/home/he/.ros/log/f3c9a586-3571-11eb-9154-484d7ea54583/kinect2-1.log].
log file: /home/he/.ros/log/f3c9a586-3571-11eb-9154-484d7ea54583/kinect2-1*.log
[kinect2_bridge-2] process has died [pid 16111, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/home/he/.ros/log/f3c9a586-3571-11eb-9154-484d7ea54583/kinect2_bridge-2.log].
log file: /home/he/.ros/log/f3c9a586-3571-11eb-9154-484d7ea54583/kinect2_bridge-2*.log
[kinect2_bridge-2] restarting process
process[kinect2_bridge-2]: started with pid [18052]
[ INFO] [1607013304.184062845]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1607013304.191929676]: waitForService: Service [/kinect2/load_nodelet] could not connect to host [he-OptiPlex-5040:54153], waiting...
[kinect2_points_xyzrgb_sd-3] process has died [pid 16112, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/sd/camera_info rgb/image_rect_color:=kinect2/sd/image_color_rect depth_registered/image_rect:=kinect2/sd/image_depth_rect depth_registered/points:=kinect2/sd/points __name:=kinect2_points_xyzrgb_sd __log:=/h [... cut ...] e_rect:=kinect2/qhd/image_depth_rect depth_registered/points:=kinect2/qhd/points __name:=kinect2_points_xyzrgb_qhd __log:=/home/he/.ros/log/f3c9a586-3571-11eb-9154-484d7ea54583/kinect2_points_xyzrgb_qhd-4.log].
log file: /home/he/.ros/log/f3c9a586-3571-11eb-9154-484d7ea54583/kinect2_points_xyzrgb_qhd-4*.log
[kinect2_points_xyzrgb_qhd-4] restarting process
process[kinect2_points_xyzrgb_qhd-4]: started with pid [18102]

If I open the image, I will report the errors. Am I missing some installation?

aoligeihahah, 2020-12-05:
hello@marcoesposito1988，How should I get aruco, which should publish tagged poses in TF

marcoesposito1988, 2020-12-06:
There are some errors related to the kinect2 nodelets in your log:

```
[FATAL] [1607013303.924090268]: Failed to load nodelet '/kinect2_points_xyzrgb_hdof typedepth_image_proc/point_cloud_xyzrgbto managerkinect2'
[FATAL] [1607013303.924309265]: Failed to load nodelet '/kinect2_points_xyzrgb_qhdof typedepth_image_proc/point_cloud_xyzrgbto managerkinect2'
[FATAL] [1607013303.924481145]: Failed to load nodelet '/kinect2_points_xyzrgb_sdof typedepth_image_proc/point_cloud_xyzrgbto managerkinect2'
[FATAL] [1607013303.924484701]: Failed to load nodelet '/kinect2_bridgeof typekinect2_bridge/kinect2_bridge_nodeletto managerkinect2'
[kinect2-1] process has died [pid 16110, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=kinect2 __log:=/home/he/.ros/log/f3c9a586-3571-11eb-9154-484d7ea54583/kinect2-1.log].
log file: /home/he/.ros/log/f3c9a586-3571-11eb-9154-484d7ea54583/kinect2-1*.log
[kinect2_bridge-2] process has died [pid 16111, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/home/he/.ros/log/f3c9a586-3571-11eb-9154-484d7ea54583/kinect2_bridge-2.log].
log file: /home/he/.ros/log/f3c9a586-3571-11eb-9154-484d7ea54583/kinect2_bridge-2*.log`
```

If the `kinect2_bridge` node is not running, the topic with the RGB image may not exist. In that case, the `aruco_ros` node can't work, because it is not getting any input. 

I would suggest you to check:
- that the RGB image topic is working
- that `aruco_ros` is getting the rectified RGB image as input
- that all input parameters to `aruco_ros` are correct (marker ID, marker size)
- that the detection is working (the aruco result topic that you already mentioned, and that the pose of the marker is published into `tf`)

If you can't get `aruco_ros` to work, maybe you can try [easy_aruco](https://github.com/marcoesposito1988/easy_aruco) instead.

aoligeihahah, 2020-12-10:
I have another question. You have plugins in the rqt document display, but my rqt does not. What is the reason?
[image]
[image]
I want to turn on rqt_ easy_ In the handeye menu bar, select plugins - visualization - image view. What can you do？

aoligeihahah, 2020-12-11:
hello@marcoesposito1988，This is the marker parameter I detected by aruco.
[image]
How do I get the pose of publishing tags in TF and how to use TF frame that must be inserted into tag as framework easy_ handeye/ calibrate.launch ? Can you write an example for me

wyf2015fei, 2020-12-17:
try to 
apt-get install rqt*

marcoesposito1988, 2020-12-22:
> hello@marcoesposito1988，This is the marker parameter I detected by aruco.
> [image]
> How do I get the pose of publishing tags in TF and how to use TF frame that must be inserted into tag as framework easy_ handeye/ calibrate.launch ? Can you write an example for me

@lyh458 already did; you can find it here: https://github.com/IFL-CAMP/easy_handeye/blob/master/docs/example_launch/iiwa_kinect_xtion_calibration.launch

zhangtianren, 2021-04-28:
> [image]

i have the same question :  NO MENU BAR !  i can not select plugins - visualization - image view. if you have solved it pls tell me tks

wyf2015fei, 2021-04-28:
> > [image]
> 
> i have the same question : NO MENU BAR ! i can not select plugins - visualization - image view. if you have solved it pls tell me tks

rqt_imageview you can see it

a2824256, 2021-04-28:
> > [image]
> 
> i have the same question : NO MENU BAR ! i can not select plugins - visualization - image view. if you have solved it pls tell me tks

Look at my Chinese blog to solve your problem. 
https://blog.csdn.net/a2824256/article/details/113127740

si1mari11ion, 2021-05-16:
hello @ @

> [image]
> If I click takesample, I will report an error. I don't know the reason and ask for your help
>
> Error processing request: "camera_marker" passed to lookupTransform argument source_frame does [...]
hello i encontered the same problem ,when i clicked on TAKE SAMPLE,the error comes out like above,could someone please tell me how to solve it?it has driven me  mad....
@aoligeihahah bro how did you solve that,could you plaese tell me ,thank you