IFL-CAMP/easy_handeye #71: There's a typo in use case sec pic

maggielovedd, 2020-12-11:
Thank you for this code. The typo is a bit confusing, I suppose the sec pic in use case should be eye-in-hand

marcoesposito1988, 2020-12-12:
Yes, that's right. However, this typo is already in the launch files, and I didn't find a way change that without breaking compatibility for existing users. If I fix the documentation without changing the launch files, there would be an inconsistency, which would bring probably even more confusion than the typo.

If you have a suggestion for a migration path that doesn't disturb existing users, that's more than welcome. Otherwise I will fix that in the ROS2 version directly :sweat_smile:

marcoesposito1988, 2020-12-12:
Oh, actually the documentation is already inconsistent. Well, I'll take a better look when I find the time. Thanks for reporting!