IFL-CAMP/easy_handeye #72: calibratin error range about 2-3 cm

IanYangChina, 2020-12-16:
Hi, thanks for the codes!
I have carried out the hand-on-base calibration with a kuka and a zivid camera.
I tried 3 times and got 3 slightly different results, with an error range about 2-3 cm.
I have checked other issues in this repo, and made sure I got the right marker id and size, took samples when the marker frame is steady in the Rviz simulation (it sometimes flickers).
I used the default algorithms for both aruco marker tracking and calibration.

Here's the results:
`parameters:
  eye_on_hand: false
  move_group: manipulator
  move_group_namespace: /iiwa
  namespace: /my_eye_on_base/
  robot_base_frame: iiwa_link_0
  robot_effector_frame: iiwa_link_ee
  tracking_base_frame: zivid_optical_frame
  tracking_marker_frame: camera_marker
transformation:
  qw: 0.04948431839228293
  qx: 0.006091250404755587
  qy: -0.9948842771176843
  qz: 0.08786053747718335
  x: -0.5081866403237757
  y: 0.1642988819302053
  z: 1.0733152560547734

parameters:
  eye_on_hand: false
  move_group: manipulator
  move_group_namespace: /iiwa
  namespace: /my_eye_on_base/
  robot_base_frame: iiwa_link_0
  robot_effector_frame: iiwa_link_ee
  tracking_base_frame: zivid_optical_frame
  tracking_marker_frame: camera_marker
transformation:
  qw: 0.02473297586797349
  qx: 0.0009234893264673997
  qy: -0.995647569407959
  qz: 0.0898517924373391
  x: -0.5351059892468916
  y: 0.15972440464955695
  z: 1.0835655779153843

parameters:
  eye_on_hand: false
  move_group: manipulator
  move_group_namespace: /iiwa
  namespace: /my_eye_on_base/
  robot_base_frame: iiwa_link_0
  robot_effector_frame: iiwa_link_ee
  tracking_base_frame: zivid_optical_frame
  tracking_marker_frame: camera_marker
transformation:
  qw: 0.06580536882194765
  qx: 0.009977368156501928
  qy: -0.9917885237153115
  qz: 0.1092045318907628
  x: -0.42471393222760484
  y: 0.1898217777090142
  z: 1.0544555687445611
`

So, just wanna be sure such an error is normal or not? Thanks :)

I noticed there are three arguments in the example file iiwa_kinect_xtion_calibration.launch (below). Are they necessary for calibration?
`
    <arg name="freehand_robot_movement" value="false" />
    <arg name="robot_velocity_scaling" value="0.5" />
    <arg name="robot_acceleration_scaling" value="0.2" />
`

IanYangChina, 2020-12-18:
I figured out that the error came from the aruco marker detection process.

closed 2020-12-18 by IanYangChina

GuoqiangRen, 2021-03-15:
> I figured out that the error came from the aruco marker detection process.

Hi, recently I met the same problem, and not sure what you mean by aruco marker detection process, can you illustrate more about this?

IanYangChina, 2021-03-15:
> > I figured out that the error came from the aruco marker detection process.
> 
> Hi, recently I met the same problem, and not sure what you mean by aruco marker detection process, can you illustrate more about this?

So, the eye-to-hand calibration needs an aruco marker attached on the end-effector. The camera will use the aruco-ros package to estimate the pose of the aruco marker relative to the camera frame. That detection process may have some error.

Beside that, another error source is the end-effector link poses read out from the robot, If the robot's own state computation process is not accurate.

lin-fangzhou, 2024-01-11:
I calibrated the robotic arm, but each time there were huge differences in the results. One noteworthy value is that my camera was always fixed on my hand, and I calibrated it 6 times as follows:
result 1
translation:
 x: 0.17973868323082823
 y: 0.006782370501410505
 z: 0.07438441881749708
rotation:
 x: -0.45153815132695097
 y: 0.5453857381574472
 z: -0.43437538148312277
 w: 0.5567636145368047

result 2
translation:
 x: -0.06653427597392521
 y: 0.025643998598674816
 z: 0.06696783370561568
rotation:
 x: -0.5327081554416593
 y: 0.5008184543910814
 z: -0.46062857442271155
 w: 0.5032138842406616

result 3
translation:
 x: 0.0769910508819065
 y: 0.12000227049890966
 z: 0.02890224129850164
rotation:
 x: -0.506780500127259
 y: 0.4405819116710794
 z: -0.6227926098119726
 w: 0.4014853284525028

result 4
translation:
 x: -0.010360640808195888
 y: 0.0058058783075903
 z: 0.07165832040066378
rotation:
 x: -0.4284056687898974
 y: 0.5838619571409756
 z: -0.382383827061095
 w: 0.5738958152437608

result 5
translation:
 x: 0.05797963988648782
 y: 0.06623176322982123
 z: 0.021080226239054548
rotation:
 x: -0.4491078296992648
 y: 0.5201030730048655
 z: -0.5053875608486993
 w: 0.5218988063725695

result 6
translation:
 x: 0.01822461983207547
 y: 0.023970008797324964
 z: 0.05262544780092089
rotation:
 x: -0.5289914513286127
 y: 0.5074252851659499
 z: -0.4647529069925332
 w: 0.4966813463756137

TowoC, 2025-01-22:
> > > I figured out that the error came from the aruco marker detection process.
> > 
> > 
> > Hi, recently I met the same problem, and not sure what you mean by aruco marker detection process, can you illustrate more about this?
> 
> So, the eye-to-hand calibration needs a marker attached on the end-effector. The camera will use the estimated pose of the marker relative to the camera frame. That detection process may have some error. In my case, I use an aruco marker, you can use other markers, such as a checkerboard.
> 
> Beside that, another error source is the end-effector link poses read out from the robot, If the robot's own state computation process is not accurate. This may happen due to wear-and-tear of the robot arm.

Hi @IanYangChina, I've met same problem as your situation. And I wonder to know the meaning that you say: the error came from the aruco marker detection process.

I also use aruco marker as the target of detection. And the error I got is about 1~2 cm. I want to know that how did you solve the problem of the error? Thanks!