IFL-CAMP/easy_handeye #73: is it possible to disable Moveit!

jackie-wayland, 2020-12-23:
can we run this calibration without move_group?

marcoesposito1988, 2020-12-23:
Hi @jackie-wayland,

yes, you can simply set the `freehand_robot_movement` argument of `calibrate.launch` to true (see [here](https://github.com/IFL-CAMP/easy_handeye/blob/master/easy_handeye/launch/calibrate.launch#L11)). 

But: you will still need the robot to publish its _updated_ pose into `tf` somehow. Please check that moving the robot makes the displayed position move in RViz before starting.

jackie-wayland, 2020-12-23:
@marcoesposito1988 after set `freehand_robot_movement` as true, the `robot.py` still complain the `move_group` not found .

https://github.com/IFL-CAMP/easy_handeye/blob/e9ecd335202fb17030664479cb1e79807617c977/easy_handeye/launch/calibrate.launch#L48

Could you fix this?

marcoesposito1988, 2020-12-23:
Sure, `robot.py` was introduced in a recent refactoring and I forgot to exclude it. Sorry about that.

What problem do you get with `rqt_easy_handeye` exactly?

jackie-wayland, 2020-12-27:
@marcoesposito1988  the `rqt_easy_handeye` can't get the correct namespace.

marcoesposito1988, 2021-02-25:
Was this the same problem as in #77? It should be solved by now then.

jackie-wayland, 2021-02-27:
@marcoesposito1988  I will test it.

closed 2021-11-21 by marcoesposito1988