IFL-CAMP/easy_handeye #77: Stuck at loading parameters

AlexSunNik, 2021-01-12:
I disabled the MoveIt and RViz but checked beforehand that all the required tf frames will be provided and updated properly. However, when running the launch file, I'm stuck at:
[INFO] [1610429294.391226]: Loading parameters for calibration /my_eob_calib_eye_on_base/ from the parameters server

Any idea why?
Thanks again for the amazing work and help.

AlexSunNik, 2021-01-13:
Problem Solved. You should do the motion part of the function `set_namespace()` in `handeye_client.py` based on the `freehand_robot_movement` argument. After I make the changes, I'm able to do the calibration now. The result computed seems reasonable by just visualizing it in Rviz. Thanks again for the amazing work.

marcoesposito1988, 2021-02-14:
I see.. thanks for the bug report!

marcoesposito1988, 2021-02-14:
This should be now fixed with version 0.4.2

closed 2021-06-20 by marcoesposito1988