IFL-CAMP/easy_handeye #79: Robot semantic description not found. Did you forget to define or remap '/robot_description_semantic'?

aycameric, 2021-01-30:
Hi,

I also disabled moveit but I get this error:  Robot semantic description not found. Did you forget to define or remap '/robot_description_semantic'?
And my robot model is not correct in rviz, why is that?
Thanks!
Best regards

marcoesposito1988, 2021-02-14:
Hi @aycameric,

without MoveIt, you need some other way of connecting the robot to ROS, so that the current pose is published in `tf`. Few robots actually have a way of doing this, so you may be better off using MoveIt.

If you do so, the `*_description` package for your robot will be automatically launched to store the URDF of the robot on the ROS parameter server. The `*_moveit_config` package should store the SRDF into the `/robot_description_semantic` parameter. This will avoid the problem you are encountering.

aycameric, 2021-02-25:
Hi **@marcoesposito1988,**
I was publishing the current pose in tf in teaching mode.

marcoesposito1988, 2021-02-25:
Perfect, this will allow you to calibrate.

However, unless the pose of each link is published (and not just the end effector pose, as I presume), the robot visualization can't work. Usually this is done by the `robot_state_publisher`, which listens to the `/joint_states` topic and does the forward kinematic. RViz then just renders each link's mesh at the corresponding tf pose, according to the URDF.

The message you are seeing is the MoveIt RViz plugin complaining, because you didn't launch MoveIt and hence the URDF and SRDF were not loaded on the ROS parameter server. You can ignore that. Everything will work, if you take care that you are going to publish the hand-eye calibration with respect to the same end-effector frame (the tf publisher in teaching mode may use a different frame than MoveIt; you just have to check that).

closed 2021-12-03 by marcoesposito1988