IFL-CAMP/easy_handeye #80: Robot base position if robot is not on the ground

ranjitkathiriya, 2021-05-07:
Hello,

I am using the easy_handeye package for my robot. I am using eye-in-hand as a calibration. 

I have a question:

1. As describe in a screenshot my robot_base and world position id different. I should use which position as a robot_base_frame? It should be robot_base or world?

`<launch>
  
  

  <include file="$(find easy_handeye)/launch/calibrate.launch">
    <arg name="eye_on_hand" value="true"/>

    
    <arg name="namespace_prefix" value="my_eih_calib"/>

    
    <arg name="robot_base_frame" value="/base_link"/>
    <arg name="robot_effector_frame" value="/ee_link"/>

    
    <arg name="tracking_base_frame" value="/optical_origin"/>
    <arg name="tracking_marker_frame" value="/optical_target"/>
  </include>
</launch>`

[image]

marcoesposito1988, 2021-05-10:
Hi @ranjitkathiriya,

if this is an eye-in-hand calibration, the base frame is indifferent. The result of the calibration will be between the chosen end effector frame and the camera frame. 

Just for reference, if it were an eye-on-base calibration, it still wouldn't matter as long as you are using `easy_handeye`'s publisher. Otherwise you would have to pay attention to the frames you chose when transferring the resulting transformation to the other program.

ranjitkathiriya, 2021-05-14:
> if this is an eye-in-hand calibration, the base frame is indifferent. The result of the calibration will be between the chosen end effector frame and the camera frame.

Yes, I am using an eye-in hand. Thanks! I got it.

closed 2021-05-14 by ranjitkathiriya