IFL-CAMP/easy_handeye #81: click on "Take Sample" Error

si1mari11ion, 2021-05-13:
Hello, I just set all the environment for  using easy_handeye,my hardwares are kinova jaco2 and intel realsense d435i.when i click on the "Take sample"button,the rqt interface become frozen for about several seconds,and then flashed off(away) ,vanished,at the same time there are error on the terminal as shown in the image 
[image]
I tried to install some corresbonding  ros packages as much as I could think of,but it didn't work,the problem persists.
Could you please give me some sulotions on it? I have drained my brain to figure it out but nothing good comes out.

si1mari11ion, 2021-05-14:
@marcoesposito1988 HI could you please give me some advice on the problem,I feel really dazed on it.

lyh458, 2021-05-14:
This issue may similar to #51. 
> is the marker visible to the camera, and detected? Working marker tracking is a prerequisite for running the calibration.

Simply, just check if you can get a result when subscribe the topic ``subscribe`` with rqt_image_viewer.

marcoesposito1988, 2021-05-14:
Yes, this is due to the fact that the tf reference frame `camera_marker` doesn't seem to exist. 

If tracking is working, it is probably assigning a different name to the marker frame, and you will need it to provide the correct name as input to the `easy_handeye` launch file, through the `tracking_marker_frame` parameter.

Simon-Pickle, 2021-05-16:
I just solved the problem and now I find any initial pose of the robot works well after clicking the button of taking a sample.
My solution was referred one of the answers from the last post and you should try to add a Image View plugin to your rqt GUI.
Then you should add the topic like aruco/result and then you can see the marker with calibration cordinate in your image view. At this time, try to click taking a sample and it won't disappaear and more, instead it shows the current frame value in the rqt GUI. Then every proceedure is as told before.

si1mari11ion, 2021-07-05:
thank you for send the solution to me,I will try it to see if it can fix the problem.thanks

zxl5344, 2021-11-18:
I got same issue. I use rqt_image_viewer and marker can be detected. 

When I take sample, I got error:
[ERROR] [1637255720.633115]: Error processing request: "camera_marker" passed to lookupTransform argument source_frame does not exist. 
['Traceback (most recent call last):\n', '  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_service.py", line 633, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/home/zxl5344/IROS2022/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_server.py", line 88, in take_sample\n    self.sampler.take_sample()\n', '  File "/home/zxl5344/IROS2022/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_sampler.py", line 88, in take_sample\n    transforms = self._get_transforms()\n', '  File "/home/zxl5344/IROS2022/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_sampler.py", line 76, in _get_transforms\n    opt = self.tfBuffer.lookup_transform(self.handeye_parameters.tracking_base_frame,\n', '  File "/opt/ros/noetic/lib/python3/dist-packages/tf2_ros/buffer.py", line 86, in lookup_transform\n    return self.lookup_transform_core(target_frame, source_frame, time)\n', 'tf2.LookupException: "camera_marker" passed to lookupTransform argument source_frame does not exist. \n']

Here is my launch file:
<launch>
	<arg name="namespace_prefix" default="iiwa_rsL515_handeyecalibration" />
	<arg name="eye_on_hand" default="false" />

    <arg name="move_group_namespace" value="/iiwa" />
    <arg name="move_group" value="manipulator" />
    <arg name="robot_base_frame" value="iiwa_link_0" />
    <arg name="robot_effector_frame" value="iiwa_link_ee" />
    <arg name="tracking_base_frame" value="tracking_origin" />
    <arg name="tracking_marker_frame" value="tracking_marker" />
    <arg name="freehand_robot_movement" default="False" />

	
    
    <include file="$(find realsense2_camera)/launch/rs_camera.launch"> </include>
	<rosparam>
      /camera/motion_module/global_time_enabled: true
      /camera/l500_depth_sensor/global_time_enabled: true
      /camera/rgb_camera/global_time_enabled: true
    </rosparam>

        
        <arg name="markerId"        default="26"/>      
        <arg name="markerSize"      default="0.1"/>    
        <arg name="marker_frame"    default="aruco_marker_frame"/> 
        <arg name="ref_frame"       default=""/>  
        <arg name="camera_frame" default="camera_color_frame" />
        <arg name="camera_image_topic" default="/camera/color/image_raw" />
        <arg name="camera_info_topic" default="/camera/color/camera_info" /> 

    <node pkg="aruco_ros" type="single" name="aruco_single">
        <remap to="$(arg camera_info_topic)" from="/camera_info" />
        <remap to="$(arg camera_image_topic)" from="/image" />
        <param name="image_is_rectified" value="True"/>
        <param name="marker_size"        value="$(arg markerSize)"/>
        <param name="marker_id"          value="$(arg markerId)"/>
        <param name="reference_frame"    value="$(arg ref_frame)"/>   
        <param name="camera_frame"       value="$(arg camera_frame)"/>
        <param name="marker_frame"       value="$(arg marker_frame)" />
    </node>
	
	
    <include file="$(find iiwa_moveit)/launch/demo.launch">
        <arg name="driver" value="True" />
        <arg name="gui" value="False" />
    </include>
	
	
    <include file="$(find easy_handeye)/launch/calibrate.launch" >
        <arg name="namespace_prefix" value="$(arg namespace_prefix)" />
        <arg name="eye_on_hand" value="false" />
        <arg name="start_rviz" value="true"/>
        <arg name="move_group_namespace" value="/iiwa" />
        <arg name="move_group" value="manipulator" />
        
        <arg name="rviz_config_file" value="$(find easy_handeye)/launch/iiwa_stack_config.rviz" />
        <arg name="tracking_base_frame" value="camera_link" />
        <arg name="tracking_marker_frame" value="camera_marker" />
        <arg name="robot_base_frame" value="iiwa_link_0" />
        <arg name="robot_effector_frame" value="iiwa_link_ee" />
        <arg name="freehand_robot_movement" value="false" />
        <arg name="robot_velocity_scaling" value="0.5" />
        <arg name="robot_acceleration_scaling" value="0.2" />
    </include>
</launch>

marcoesposito1988, 2021-11-21:
@zxl5344 unless I'm missing something, this can't work. Instead of sampling the output of the marker tracking, the transform from `camera_link` to `camera_color_frame` will be sampled, independently of the position of the robot.

You should pass `$(arg marker_frame)` as `tracking_marker_frame` to `$(find easy_handeye)/launch/calibrate.launch`

zxl5344, 2021-11-22:
@marcoesposito1988 Thanks for pointing out my mistake. I have fixed it.

closed 2021-11-29 by marcoesposito1988