IFL-CAMP/easy_handeye #83: Can this package be used to calibrate a camera mounted on a two-axis gimbal?

huakang402, 2021-05-16:
Hello!
Can this package be used to calibrate a camera mounted on a two-axis gimbal? If so, what should I pay attention to?
[image]

I tried use pitch as robot_effector_frame and Apriltag to generate tracking_marker_frame, but the calibration result is not very good. This is the parameter I set.
[image]

marcoesposito1988, 2021-05-17:
Hi @huakang402,

unfortunately I would be surprised if you can use a standard hand-eye calibration algorithm with a 2DoF robot. My intuition is that the set of acquired poses will not sufficiently constrain the solution space, and the resulting system of equations will be underdetermined. I would be happy to be proven wrong, of course.

Dedicated approaches may exist, but I am not familiar with any. You can try researching the literature. You can then plug the algorithm into easy_handeye and send a PR, if you succeed!

huakang402, 2021-05-17:
Hi @marcoesposito1988,
I used the chessboard to calibrate the camera again, and I successfully used easy_handeye to calibrate the hand-eye between the gimbal and the camera!
Thank you so much for the easy_handeye!

marcoesposito1988, 2021-05-18:
That's great news, I'm happy to hear that!

closed 2021-05-31 by marcoesposito1988