IFL-CAMP/easy_handeye #84: I got an error when click take sample

irenjie, 2021-05-30:
Error processing request: Lookup would require extrapolation into the past.  Requested time 1622358795.871479988 but the earliest data is at time 1622358795.983241320, when looking up transform from frame [aruco_marker_frame] to frame [camera_link]
['Traceback (most recent call last):\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 625, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/home/ma/Documents/ros_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_server.py", line 88, in take_sample\n    self.sampler.take_sample()\n', '  File "/home/ma/Documents/ros_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_sampler.py", line 88, in take_sample\n    transforms = self._get_transforms()\n', '  File "/home/ma/Documents/ros_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_sampler.py", line 78, in _get_transforms\n    Duration(10))\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_ros/buffer.py", line 87, in lookup_transform\n    return self.lookup_transform_core(target_frame, source_frame, time)\n', 'ExtrapolationException: Lookup would require extrapolation into the past.  Requested time 1622358795.871479988 but the earliest data is at time 1622358795.983241320, when looking up transform from frame [aruco_marker_frame] to frame [camera_link]\n']
Traceback (most recent call last):
  File "/home/ma/Documents/ros_ws/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_easy_handeye.py", line 132, in handle_take_sample
    sample_list = self.client.take_sample()
  File "/home/ma/Documents/ros_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_client.py", line 76, in take_sample
    return self.take_sample_proxy().samples
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 515, in call
    responses = transport.receive_once()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 728, in receive_once
    p.read_messages(b, msg_queue, sock) 
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 353, in read_messages
    self._read_ok_byte(b, sock)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 336, in _read_ok_byte
    raise ServiceException("service [%s] responded with an error: %s"%(self.resolved_name, str))
rospy.service.ServiceException: service [/aubo_kinect_handeyecalibration_eye_on_hand/take_sample] responded with an error: error processing request: Lookup would require extrapolation into the past.  Requested time 1622358795.871479988 but the earliest data is at time 1622358795.983241320, when looking up transform from frame [aruco_marker_frame] to frame [camera_link]
[aubo_kinect_handeyecalibration_eye_on_hand/namespace_ubuntu_56042_8581700767765726530_rqt-4] process has died [pid 56077, exit code 1, cmd /home/ma/Documents/ros_ws/src/easy_handeye/rqt_easy_handeye/scripts/rqt_easy_handeye __name:=namespace_ubuntu_56042_8581700767765726530_rqt __log:=/home/ma/.ros/log/45258028-c116-11eb-b354-000c29f15f06/aubo_kinect_handeyecalibration_eye_on_hand-namespace_ubuntu_56042_8581700767765726530_rqt-4.log].
log file: /home/ma/.ros/log/45258028-c116-11eb-b354-000c29f15f06/aubo_kinect_handeyecalibration_eye_on_hand-namespace_ubuntu_56042_8581700767765726530_rqt-4*.log

irenjie, 2021-05-30:
I have solved this error by set the set the  'reference_frame' in single.launch

closed 2021-05-30 by irenjie

Sandrolek, 2022-02-25:
Hello, where did you find file single.launch?

Sandrolek, 2022-03-02:
> > Hello, where did you find file single.launch?
> 
> Hi @Sandrolek,
> 
> single.launch is part of aruco_ros https://github.com/pal-robotics/aruco_ros/blob/noetic-devel/aruco_ros/launch/single.launch

Oh, thanks a lot!